MarlinFirmware/Marlin #25730: MKS Robin Nano V3.0 - MKS Wifi - file management issue

pagabee, 2023-04-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I saw that, there are updates in the _mks_ui_ folder, so I tried it out and found some issues regarding the file management.

### Bug Timeline

New bug, no problem with Marlin 2.1.2

### Expected behavior

I expect the same behaviour as Marlin 2.1.2

[image]

### Actual behavior

The files are not deleted and the printing buttons are missing.
There are _0_ characters at the end of the filenames.

[image]

### Steps to Reproduce

I'm using an old version of ESP3D V3.0, because the upload speed is higher then in the newer versions.

[image]

### Version of Marlin Firmware

Marlin bugfix 2.1.x - 21.04.2023

### Printer model

Custom

### Electronics

MKS Robin Nano V3.0,

### Add-ons

MKS TS35 V2.0, MKS Wifi with ESP3D v3.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/11301009/Configs.zip)

ellensp, 2023-04-23:
LONG_FILENAME_HOST_SUPPORT used to also add the size to the M20 output on the older marlin (even without L parameter)
 
But now the length is always displayed 

This was changed in https://github.com/MarlinFirmware/Marlin/commit/1bd921d6a69412686fa4b31c4c1e5710de9840cf#diff-12f5e0057d5add14fc34e3f8b1efa9ed52245b3a31c73b7665c45274ab0ca51f

So the wifi code would need to be updated to understand this

or you can remove the size from marlin
comment out this line https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/sd/cardreader.cpp#L326

pagabee, 2023-04-23:
This was modified on Jan 6, 2022: https://github.com/MarlinFirmware/Marlin/commit/1bd921d6a69412686fa4b31c4c1e5710de9840cf#diff-12f5e0057d5add14fc34e3f8b1efa9ed52245b3a31c73b7665c45274ab0ca51f
Marlin 2.1.2 was released on Dec 19, 2022.

I checked the _M20_ and _M20 L_ G-codes via USB-Serial port (not trough MKS Wifi). The responses are the same for both Marlin 2.1.2 and the actual Marlin 2.1.x:

[image]

Therefore I think that the issue is related to the _mks_ui_ folder. The MKS Wifi Serial is handeld by the files of this folder.

closed 2023-07-03 by github-actions[bot]