MarlinFirmware/Marlin #25733: [BUG] Ender 3 Pro Stock Part Cooling Fan Does Not Run below 100%

sarvenn, 2023-04-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Ender 3 Pro Stock Part Cooling Fan Does Not Run below 100%. It stops spinning at 99%. So, it only runs at full speed. Whenever I drop the fan speed to 99%, fan stops spinning.
It was working properly on Bugfix 2.0. I suppose it was also running properly on previous Bugfix 2.1s.

You may find my config files attached. I do not change any parameter related to fan on the default config files.

I use BTT SKR mini E3 v2 board with TFT35.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11303417/Configuration.zip)

### Bug Timeline

Bugfix 2.1

### Expected behavior

Fan should spin at least at 50%.

### Actual behavior

Fan does not spin below 100% fan speed.

### Steps to Reproduce

Set fan speed to 100%. Fan spins. Drop to 99%. Fan stops.
Set fan speed to 99%. Fan does not spin. Set to 100%, fan spins at full speed.

### Version of Marlin Firmware

Recent Bugfix 2.1

### Printer model

Ender 3 Pro

### Electronics

SKR Mini E3 V2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-23:
enable #define FAN_SOFT_PWM

sarvenn, 2023-04-23:
OK,I will, but what I don't understand is that this parameter was undefined at release marlin 2.0.x and bugfix versions and I did not have this issue.
In the past, I remember that fan was not working at all but it was solved with bug reports and PRs.

thisiskeithb, 2023-04-27:
> It was working properly on Bugfix 2.0. I suppose it was also running properly on previous Bugfix 2.1s

Were you compiling with the same PIO environment between versions? Which environment were you using?

I just tried the [Creality/Ender-3/BigTreeTech SKR Mini E3 2.0](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0) `bugfix-2.1.x` config with the `STM32F103RC_btt` environment using a stock, barely used Ender-3 4010 parts cooling fan & it is adjustable from 28%-100%.

The default `Creality/Ender-3/BigTreeTech SKR Mini E3 2.0` config could be improved by setting a `FAN_MIN_PWM` of `72` to bring back the full range.

With `FAN_SOFT_PWM` enabled, the parts cooling fan is adjustable from 16%-100%, so a `FAN_MIN_PWM` of `41` would help there. _Note: Bumping `SOFT_PWM_SCALE` to `1` or `2` cuts down on the "ticking" you get with `FAN_SOFT_PWM`, but `FAN_MIN_PWM` needs to be ≥57_

If your fan only runs at 100%, it could be on its way out. I also suspect that Creality has/had multiple OEMs for their fans and they work differently depending on when you purchased your printer.

sarvenn, 2023-04-28:
> > It was working properly on Bugfix 2.0. I suppose it was also running properly on previous Bugfix 2.1s
> 
> Were you compiling with the same PIO environment between versions? Which environment were you using?
> 
> I just tried the [Creality/Ender-3/BigTreeTech SKR Mini E3 2.0](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0) & [Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 2.0](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0) `bugfix-2.1.x` configs with the `STM32F103RC_btt` environment using a stock, barely used Ender-3 4010 parts cooling fan (Creality DC Brushless PB04010MS4) & it is adjustable from 28%-100%.
> 
> The default SKR Mini E3 V2 configs could be improved by setting a `FAN_MIN_PWM` of `72` to bring back the full range.
> 
> With `FAN_SOFT_PWM` enabled, the parts cooling fan is adjustable from 16%-100%, so a `FAN_MIN_PWM` of `41` would help there. _Note: Bumping `SOFT_PWM_SCALE` to `1` or `2` cuts down on the "ticking" you get with `FAN_SOFT_PWM`, but `FAN_MIN_PWM` needs to be ≥57_
> 
> If your fan only runs at 100%, it could be on its way out. I also suspect that Creality has/had multiple OEMs for their fans and they work differently depending on when you purchased your printer.

Hi,
I generally use Autobuild Marlin and always use STM32F103RC_btt (256K) environment.
That's quite weird that we both used the same configs and got different results.

I just tried in TFT and Marlin modes again and got quite confusing outcomes. First I tried in TFT mode and set the fan to 50% and it works until 20%. I switched to Marlin mode and it worked again but then it stopped working again below 100% in both TFT and Marlin modes. I will try to build a fresh firmware with Autobuild Marlin and PIO extension and try it again. I will share my results. If that doesn't work either I will play on PWM settings as you said Keith.

sarvenn, 2023-04-29:
I did the changes you said and now it is working at full range @thisiskeithb .
By the way I have the same fan with you (Creality DC Brushless PB04010MS4). I don't know what cause this but it is better to change the config files with the parameters as you said above. I don't have any experience how to make a change in the repo. Could you do it? @thisiskeithb

closed 2023-04-29 by sarvenn