MarlinFirmware/Marlin #25737: [BUG] G29 command fails to make probe reach the bed

Kurolox, 2023-04-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to generate a mesh with G29 fails when trying to probe the first point. It seems like the probe is unable to reach the bed, and this causes an emergency shutdown.

Any other usage of the probe (G28, G34...) doesn't seem to be affected by this issue. It only happens during G29 probing specifically.

### Bug Timeline

Started when I updated to `2.1.X` version. 

### Expected behavior

I expect the probe to be able to reach the bed without issuing an emergency shutdown

### Actual behavior

The probe fails to reach the bed, and issues an M114 command. See below for further logs.

### Steps to Reproduce

1. Issue a `G28` command to home the axis
2. Issue a `G29 P1` command to start probing the bed
3. Watch how it fails to do so at the first point 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Apr 23 2023 18:51:04)

### Printer model

Custom Prusa i3

### Electronics

BTT SKR 2.0 Rev B

### Add-ons

BLTouch 3.1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Output of the "Steps to reproduce" section with `M111 S247`: https://pastebin.com/tRZNjZWW

[Configuration files (bugfix-2.1.x)](https://github.com/MarlinFirmware/Marlin/files/11304342/Marlin.zip)
[Configuration files (2.1.x)](https://github.com/MarlinFirmware/Marlin/files/11304344/Marlin.zip)

Video showcasing the issue following the "Steps to reproduce" section:
https://user-images.githubusercontent.com/13695988/233854869-f124306d-b42d-4619-99f9-8360c51957d8.mp4

Kurolox, 2023-04-23:
Increasing `Z_PROBE_LOW_POINT` fixed the issue as a workaround. I believe the issue is that the calculations for the probing aren't taking into account the Z value of `NOZZLE_TO_PROBE_OFFSET`, so if said value is lower than `Z_PROBE_LOW_POINT` it will fail to probe, as showcased in this issue.

While I'm not too familiar with C++, I'd love to work into this issue and see if I can attempt to fix it myself. I've been using Marlin for quite a while and I'd love to make a contribution.

thisiskeithb, 2023-04-23:
This may be fixed in https://github.com/MarlinFirmware/Marlin/pull/25631 if you want to give it a try and report your findings there.

Kurolox, 2023-04-23:
Thanks for letting me know about that! Unfortunately I've tried it, but the same issue is still present. I'll also report it there.

jamespearson04, 2023-04-23:
Should by fixed over on my PR #25631 by commit de09495.

Kurolox, 2023-04-23:
I can confirm that the commit fixed this issue. Thanks for the effort, @jamespearson04!

closed 2023-04-23 by Kurolox