MarlinFirmware/Marlin #25740: [BUG] Use of DISABLE_DEBUG in pins_CREALITY_V4.h results in unnecessary loss of SWD

Yangff, 2023-04-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In #17719 a macro `DISABLE_DEBUG` is used in all CREALITY motherboards that includes the `src/pins/stm32f1/pins_CREALITY_V4.h` header file. This results in the loss of SWD functionality on those boards regarding the  use of PB4 as in the comment. 
Also, considering that the SWD pins of the STM32F103Rx used on boards like the V422 are PA13 and PA14 (which also includes this header), this macro appears to be defined in the incorrect location. 

### Expected behavior

Being able to use SWD on earlier boards.

### Actual behavior

···
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: Error connecting DP: cannot read IDR
```

For people who are suffering from this problem and searching this error message on V422 board.. (with a corrupted bootloader and trying to flash your board) please find a pair of tweezers or something conductive and short the P3 contact on the mianboard while programming, once you see something like this
```
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x1ba01477
Info : [stm32f1x.cpu] Cortex-M3 r1p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
```
release the pins and you should be able to program the board.

### Steps to Reproduce

connect the swd to the debug port and use whatever probe the only result you will get is it cannot find the chip.

### Version of Marlin Firmware

doesn't matter

### Printer model

Creality Ender-3 Pro

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-25:
"unnecessary" is a point of view. for all those who have added additional hardware on PA13 and or  PA14 it is very necessary.

Yangff, 2023-04-25:
> "unnecessary" is a point of view. for all those who have added additional hardware on PA13 and or PA14 it is very necessary.

This is a lost function in the default configuration which does not come with extra hardware enabled. There is no clear indication of this loss with compiled, so I am not aware of PA13/14 being repurposed in this case, and even the description on this macro is that it allows the use of PB4, which makes it even unclear. (when I recognized I lost SWD, I first tried to looking for which component is using PA13/PA14 and found nothing)

closed 2023-07-07 by github-actions[bot]