MarlinFirmware/Marlin #25742: [BUG] zero extruders/zero temp sensors fails compile

chaosmstr, 2023-04-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a MKS Robin Nano V3.1 that I am attempting to set up as a CNC.
Following the dirs from MKS (https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/wiki/Marlin_firmware) I can get a build that works just fine.
But a CNC has no extruder or temp sensors.
Using either 2.1.2 or 2.1.x bugfix gives the same error.

After setting extruders=0, temp sensors=0, and reverting all of the axis calls to { x, y, z } the build will not compile.
I don't know enough to change this code myself, so I am reporting it.

`
platformio run --silent -e mks_robin_nano_v3_usb_flash_drive
Marlin\src\inc\Warnings.cpp:67:4: warning: #warning "Your Configuration provides no method to acquire user feedback!" [-Wcpp]
   67 |   #warning "Your Configuration provides no method to acquire user feedback!"
      |    ^~~~~~~
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp: In function 'void event_handler(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp:61:21: error: 'class Planner' has no member named 'flow_percentage'
   61 |         if (planner.flow_percentage[0] < MAX_EXT_SPEED_PERCENT - uiCfg.stepPrintSpeed)
      |                     ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp:62:19: error: 'class Planner' has no member named 'flow_percentage'
   62 |           planner.flow_percentage[0] += uiCfg.stepPrintSpeed;
      |                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp:64:19: error: 'class Planner' has no member named 'flow_percentage'
   64 |           planner.flow_percentage[0] = MAX_EXT_SPEED_PERCENT;
      |                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp:65:17: error: 'class Planner' has no member named 'refresh_e_factor'
   65 |         planner.refresh_e_factor(0);
      |                 ^~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_change_speed.cpp:81:21: error: 'class Planner' has no member named 'flow_percentage'
   81 |         if (planner.flow_percentage[0] > MIN_EXT_SPEED_PERCENT + uiCfg.stepPrintSpeed)
      |                     ^~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive\src\src\lcd\extui\mks_ui\draw_change_speed.cpp.o] Error 1
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:105:19: error: 'class Planner' has no member named 'flow_percentage'
  105 |           planner.flow_percentage[0] = 100;
      |                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:106:19: error: 'class Planner' has no member named 'e_factor'
  106 |           planner.e_factor[0] = planner.flow_percentage[0] * 0.01f;
      |                   ^~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:106:41: error: 'class Planner' has no member named 'flow_percentage'
  106 |           planner.e_factor[0] = planner.flow_percentage[0] * 0.01f;
      |                                         ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:192:20: error: 'class Temperature' has no member named 'setTargetHotend'; did you mean 'degTargetHotend'?
  192 |     thermalManager.setTargetHotend(uiCfg.hotendTargetTempBak, uiCfg.extruderIndex);
      |                    ^~~~~~~~~~~~~~~
      |                    degTargetHotend
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:204:20: error: 'class Temperature' has no member named 'setTargetHotend'; did you mean 'degTargetHotend'?
  204 |     thermalManager.setTargetHotend(uiCfg.hotendTargetTempBak, uiCfg.extruderIndex);
      |                    ^~~~~~~~~~~~~~~
      |                    degTargetHotend
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive\src\src\lcd\extui\mks_ui\draw_dialog.cpp.o] Error 1
=============================================================== [FAILED] Took 244.36 seconds =============================================================== 

Environment                        Status    Duration
---------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive  FAILED    00:04:04.360
=========================================================== 1 failed, 0 succeeded in 00:04:04.360 ===========================================================
`

### Expected behavior

Setting Extruders to 0 should allow for compilation.

### Actual behavior

Failed to compile calling 'class planner' variable.

### Steps to Reproduce

set extruders to 0
set temp temp sensors to 0
set all axis calls to { x, y, z } instead of { x, y, z, e } format
compile build

### Version of Marlin Firmware

2.1.2

### Printer model

home build

### Electronics

MKS Robin Nano V3.1

### Add-ons

Robin Wifi

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[MKS_Robin_Nano_Config.zip](https://github.com/MarlinFirmware/Marlin/files/11319576/MKS_Robin_Nano_Config.zip)

ellensp, 2023-04-25:
mks_ui does not support non standard configs (like no extruders)

ellensp, 2023-04-25:
In Configuration.h the following lines are invalid.
```
#define DISABLE_Xfalse
#define DISABLE_Yfalse
#define DISABLE_Zfalse
```

ellensp, 2023-04-25:
@chaosmstr 
try this I just created https://github.com/ellensp/Marlin/tree/allow-no-E-in-TFT_LVGL_UI
Compiles,but I have no idea how the display might break