MarlinFirmware/Marlin #25746: [BUG] Board doesn't react after activating TC MAX sensors (31865, 31855, 6675) in Config

OrionLightning, 2023-04-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello everyone!

I'm experiencing really strange behavior regarding Marlin after changing config for MAX thermocouple sensor. I'm kinda newbie to Marlin Firmware, so bare in mind that i could be messing something obvious up but I'm going to describe my issue as much as i can.

So i have tried to change and compile Marlin to my needs by activating MAX 31865(**-5**) as my TEMP_SENSOR_0 instead of analog thermistor(**1**). I defined my typical SPI pins according to SDCARD pins such as this:

```
#define TEMP_0_SCK_PIN PA5
#define TEMP_0_MISO_PIN PA6
#define TEMP_0_MOSI_PIN PA7
#define TEMP_0_CS_PIN PD3 // PD3 -> E3 CS PIN for octopus
```

It compiles without problem but whenever i flash it to my **OCTOPUS v1.1**, the status pin doesn't blink and stays just on after flashing has been done. USB also doesn't work after that. Neither on PC or on Octopi(Pi 4B). So I'm assuming whatever happens in Marlin Firmware, the code just gets stuck after first initialization on the BTT Board.

Just to be sure i tried the same method on **BTT SKR v1.3** and **SKR MINI e3 V2**. The behavior is the same, so no luck there either. 

I'm using the newest bugfix firmware version but i tried it on 2.1(non bugfix) and on 2.0. Again the same issue.

If i change TEMP_SENSOR_0 to 1. it works without a problem. USB works well, status led blinks all the time.

Moreover i changed internal MAX 31865 library to official Adafruit library and then to a modded Adafruit library from [GadgetAngel.](https://github.com/GadgetAngel/Adafruit-MAX31865-V1.1.0-Mod-M.git) Same issue continues on each libraries...

**PS:** I don't have MAX 31865 module attached on my board. I just wanted to activate it in config and then proceed to implement a new custom code for an another SPI device.

### Bug Timeline

Expecting to be new

### Expected behavior

USB should be working and responding to Octopi.

### Actual behavior

Status led doesn't blink, USB connection is not recognized.

### Steps to Reproduce

1) #define TEMP_SENSOR_0 -5
2) #define SDSUPPORT
3) Define the SPI pins for TEMP_0_CS etc. in OCTOPUS pins header file.
4) Compile and flash

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

None, just electronic board itself

### Electronics

BTT Octopus v1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11330185/Configuration.zip)

[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/11330198/platformio.zip)

ellensp, 2023-04-28:
When you enable a MAX sensors , its not optional, it must be connected

OrionLightning, 2023-05-04:
Thank you @ellensp. Apparently that was my problem that i didn't have tc sensor attached to my board. it works now smoothly.

It's shame though that i cant simply activate it without sensor connected. It will be a bit tricky to develop and test the custom library for other spi devices.

closed 2023-05-04 by OrionLightning