MarlinFirmware/Marlin #25747: ENDSTOP_INTERRUPTS_FEATURE breaks BLTouch on Creality boards, possibly all STM32

EvilGremlin, 2023-04-26:
z_min_probe is stuck triggered with interripts, i think it's better to disable them at least for creality boards (including dropins)

thinkyhead, 2023-04-27:
I think I may have patched your issue @EM-III … But, we still need to investigate the cause of the original issue, if that didn't also get solved by the latest endstops patch.

So if anyone has the ability to test the latest code to see if the BLTouch / STM32 / Endstop Interrupt conflict is still happening, we can start to delve into that and consider possible causes. Generally this kind of issue involves a timer or interrupt conflict, so that would be my first area of inquiry. I have a couple of spare BLTouch probes and some loose STM32 boards, so I can hook up an ST-Link and get down with some direct debugging, if the issue isn't clear just by looking at the datasheets and our endstop interrupt code.

I have other tasks to deal with today, but we can start gathering information. First, does the issue still exist? And, what are the steps to reproduce it? …preferably with one of the boards I have handy here. Of course, the larger question is whether it's an unavoidable conflict, so endstop interrupts just have to remain disabled, or is it actually solvable?

So, let's gather all the data we can about which pins and timers and interrupts are involved, and then we'll be at a good starting point to start into testing, experiments, and debugging.

EvilGremlin, 2023-04-27:
Sounds like return of that dumb bug with preprocessor going blind.

thisiskeithb, 2023-04-27:
> Sounds like return of that dumb bug with preprocessor going blind.

I remember trying to track this down and could never confirm that it was a real issue.

...Or it was a very short-lived bug.

I enable `ENDSTOP_INTERRUPTS_FEATURE` where I can on my builds / forks that I support and it seems to work fine without causing issues, but I may not be running the perfect set of hardware to reproduce it.

thinkyhead, 2023-05-14:
I suspect this is all set and can likely be closed.

closed 2024-04-16 by thisiskeithb