MarlinFirmware/Marlin #25751: [BUG] Calibration of delta impossible since #25748 (Bump 27-04-2023)

Foxies-CSTL, 2023-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I use a removable probe under the nozzle and when starting delta calibration, the head takes a 1st measurement but does not come up and moves to take the 2nd measurement, then the 3rd measurement and the head lifts diagonally to reach the 4th point. The head lowers to take the measurement but does not reach the bed. It ends up at an undefined height with a "probing failed" message.

### Version of Marlin Firmware

bugfix 2.1.3 bump 20230427

### Printer model

Delta

### Electronics

Nanov3.1

### Add-ons

removable probe

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

`Send: G33 V3
Recv: G33 Auto Calibrate
Recv: Checking... AC at radius:117.0000
Recv: .Height:330.0000  Ex:0.0000  Ey:0.0000  Ez:0.0000  Radius:151.6700
Recv: .               Tx:0.0000  Ty:0.0000  Tz:0.0000
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: Deploy Z-Probe
Recv: //action:prompt_end
Recv: //action:prompt_begin Deploy Z-Probe
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
[...]
Send: M876 S0
Recv: Wait for bed heating...
[...]
Recv:  T:-1.00 /0.00 (4089.1875) B:59.63 /60.00 (3431.5000) M:26.23 (3894.1875) @:0 B@:91
Recv:  T:-1.00 /0.00 (4088.0625) B:59.84 /60.00 (3427.1875) M:26.41 (3892.7500) @:0 B@:76
[...]
Recv: Stow Z-Probe
Recv: //action:prompt_end
Recv: //action:prompt_begin Stow Z-Probe
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
[...]
Send: M876 S0
Recv: Error:Probing Failed
Recv: Correct delta settings with M665 and M666`

thisiskeithb, 2023-04-28:
Please try PR https://github.com/MarlinFirmware/Marlin/pull/25631 and report your findings there.

Foxies-CSTL, 2023-04-28:
I made a patch application on my BugFix sources (27/04/2023) and it doesn't work. I worked from the PR #25631 and activated the debug mode. Same observation. Add a text file with M503,M115,M111 S32. 
[Delta-PR#25631.txt](https://github.com/MarlinFirmware/Marlin/files/11352988/Delta-PR.25631.txt)

jamespearson04, 2023-04-28:
The json won't extract from your zip, and the actual .h files would be more helpful. I may as well ask though, do you have `Z_PROBE_LOW_POINT` set? and what is your `NOZZLE_TO_PROBE_OFFSET`?

thisiskeithb, 2023-04-28:
> The json won't extract from your zip

<details><summary>marlin_config.json</summary>
<p>

```json
{
  "__INITIAL_HASH": "7a0fafdece07bc7e903e",
  "Configuration.h": {
    "VALIDATE_HOMING_ENDSTOPS": "",
    "DELTA_DIAGONAL_ROD": "315.0",
    "Y_DRIVER_TYPE": "DRIVER_AXES",
    "TEMP_SENSOR_PROBE": "0",
    "PREHEAT_1_LABEL": "\"PLA\"",
    "EXTRUDE_MAXLENGTH": "900",
    "MPC_STEADYSTATE": "0.5f",
    "TEMP_WINDOW": "1",
    "EEPROM_SETTINGS": "",
    "MPC_MAX": "BANG_MAX",
    "SEGMENT_LEVELED_MOVES": "",
    "PREHEAT_1_TEMP_HOTEND": "205",
    "PREHEAT_5_TEMP_HOTEND": "230",
    "LCD_INFO_SCREEN_STYLE": "0",
    "DISPLAY_CHARSET_HD44780": "WESTERN",
    "GRID_MAX_POINTS_Y": "GRID_MAX_POINTS_X",
    "CUSTOM_VERSION_FILE": "Version.h",
    "MPC_AUTOTUNE": "",
    "THERMAL_PROTECTION_BED": "",
    "CONFIGURATION_H_VERSION": "02010300",
    "XYZ_BELT_PITCH": "2",
    "SOFT_PWM_SCALE": "0",
    "NEOPIXEL_PIXELS": "32",
    "DEFAULT_bedKi": "52.5401",
    "PREHEAT_1_TEMP_CHAMBER": "35",
    "DUMMY_THERMISTOR_998_VALUE": "25",
    "FAN_SOFT_PWM": "",
    "Z_MAX_POS": "MANUAL_Z_HOME_POS",
    "Z_CLEARANCE_MULTI_PROBE": "5",
    "E0_DRIVER_TYPE": "DRIVER_EXT",
    "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN": "",
    "G26_XY_FEEDRATE_TRAVEL": "100",
    "MOTHERBOARD": "BOARD_MKS_ROBIN_NANO_V3_1",
    "DEFAULT_TRAVEL_ACCELERATION": "1500",
    "NOZZLE_TO_PROBE_OFFSET": "{ X_OFFSET, Y_OFFSET, Z_OFFSET }",
    "PREHEAT_5_LABEL": "\"TPU\"",
    "FIL_RUNOUT_ENABLED_DEFAULT": "false",
    "X_MIN_POS": "-(PRINTABLE_RADIUS)",
    "Z_MIN_PROBE_REPEATABILITY_TEST": "",
    "Z_PROBE_END_SCRIPT": "\"G28\"",
    "USE_ZMIN_PLUG": "",
    "MANUAL_Z_HOME_POS": "DELTA_HEIGHT",
    "PREHEAT_3_LABEL": "\"ABS\"",
    "PREHEAT_2_TEMP_CHAMBER": "35",
    "COOLER_OVERSHOOT": "2",
    "HOST_KEEPALIVE_FEATURE": "",
    "Z_PROBE_FEEDRATE_SLOW": "(Z_PROBE_FEEDRATE_FAST / 8)",
    "XY_PROBE_FEEDRATE": "(66*60)",
    "INVERT_E0_DIR": "false",
    "X_MAX_ENDSTOP_HIT_STATE": "HIGH",
    "USE_ZMAX_PLUG": "",
    "Y_MAX_ENDSTOP_HIT_STATE": "HIGH",
    "PREHEAT_3_TEMP_CHAMBER": "35",
    "TEMP_SENSOR_0": "66",
    "TEMP_SENSOR_BOARD": "0",
    "Z_MAX_ENDSTOP_HIT_STATE": "HIGH",
    "TEMP_SENSOR_BED": "11",
    "DISABLE_OTHER_EXTRUDERS": "",
    "UBL_MESH_WIZARD": "",
    "TEMP_BED_WINDOW": "1",
    "DEFAULT_RETRACT_ACCELERATION": "400",
[... 457 lines cut ...]
    "DEFAULT_STEPPER_TIMEOUT_SEC": "300",
    "FILAMENT_CHANGE_UNLOAD_LENGTH": "100",
    "TEMP_SENSOR_AD595_GAIN": "1.0",
    "FILAMENT_UNLOAD_PURGE_DELAY": "10000",
    "E2_HYBRID_THRESHOLD": "30",
    "MAIN_MENU_ITEM_8_GCODE": "\"G29L2\\nM1004B80S2\"",
    "MAIN_MENU_ITEM_4_GCODE": "\"G28\"",
    "RX_BUFFER_SIZE": "1024",
    "MAIN_MENU_ITEM_9_CONFIRM": "",
    "Y_HYBRID_THRESHOLD": "150",
    "MIN_AUTORETRACT": "0.1",
    "MAIN_MENU_ITEM_6_GCODE": "\"M306T\\nM500\\nG28\\nM107\"",
    "E7_HYBRID_THRESHOLD": "30",
    "E0_AUTO_FAN_PIN": "PB0"
  },
  "VERSION": {
    "DETAILED_BUILD_VERSION": "SHORT_BUILD_VERSION",
    "STRING_DISTRIBUTION_DATE": "\"2023-04-20\"",
    "GIT_REF": "aa375299b5"
  }
}
```

</p>
</details>

Foxies-CSTL, 2023-04-28:
Thanks @thisiskeithb

jamespearson04, 2023-04-28:
Perfect, thanks @thisiskeithb. Not sure why that didn't want to unzip for me then, even redownloaded.

Hard to figure it all out, as I don't have a delta to test on, but some of the issues are due to G33 using Probe.h functions without homing the bed/setting the delta height. This is what causes the lack of lifts from the bed on the initial points, as the nozzle thinks its 14mm higher off the bed than it actually is, and as a result thinks it's above the clearance height.

@Foxies-CSTL I've added a couple of commits to #25631 that should make the nozzle lift after probing as it should, it would be worth trying that PR again, and uploading another debug log. Roughly how high is it lifting for that fourth point? The log suggests there shouldn't really be more than 5-10mm of height variation throughout.

Foxies-CSTL, 2023-04-28:
@jamespearson04 , Ok I make a test.

Foxies-CSTL, 2023-04-28:
Same issue . I add a video.

https://user-images.githubusercontent.com/24864147/235245487-d749b3da-a336-4af0-a33a-240880a7feca.mp4

jamespearson04, 2023-04-28:
@Foxies-CSTL Thanks, the video is very helpful, could I have a fresh log to go with it too? need to see if heights have changed after my changes

Foxies-CSTL, 2023-04-28:
@jamespearson04 
The Delta height did not change:
`Recv: echo:  M665 L315.0000 R151.6700 H330.0000 S160.0000 X0.0000 Y0.0000 Z0.0000 A0.0000 B0.0000 C0.0000`
[Delta2-PR#25631.txt](https://github.com/MarlinFirmware/Marlin/files/11357044/Delta2-PR.25631.txt)

Foxies-CSTL, 2023-04-28:
Rebuilt with your last commits
[image]
=> Success!

Foxies-CSTL, 2023-04-28:
G33 Px V3=> Ok, the different Delta calibrations are working.
Wizard Probe => Ok, but I need to write a PR to disable the EndStop with the launch otherwise we can't adjust the Z_OffSet when it is below 0 (negative values).
M1004 B60 S1 => Ok, the UBL leveling by the M1004 macro works too.
Test to print the flow for sequential bridging (SuperSlicer test) on the whole bed surface => success.

[image]

Thanks for your support!

closed 2023-04-28 by thinkyhead