MarlinFirmware/Marlin #25752: [BUG] Adding a Controller fan causes compilation error for LPC1769 BOARD_MKS_SGEN_L_V2

rondlh, 2023-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After updating to the latest bugfix-2.1.x 2023-04-27 I got compilation errors when enabling USE_CONTROLLER_FAN.
I got the following errors
undefined reference to `GcodeSuite::M106()
undefined reference to `GcodeSuite::M107()

### Bug Timeline

New bug as of 2023-04-27

### Expected behavior

Compile like it used to before 2023-04-27

### Actual behavior

Compilation errors:
undefined reference to `GcodeSuite::M106()
undefined reference to `GcodeSuite::M107()

### Steps to Reproduce

1. Start from the latest bugfix (2023-04-27)
2. In platformio.ini set "default_envs = LPC1769"
3. In configuration.h set the MOTHERBOARD to BOARD_MKS_SGEN_L_V2
4. In configuration_adv.h enable USE_CONTROLLER_FAN
5. In configuration_adv.h set "#define CONTROLLER_FAN_PIN P2_06"
6. Compile... then I get:
```
undefined reference to `GcodeSuite::M106()
undefined reference to `GcodeSuite::M107()
```
7. If I add "#define HAS_FAN 1" then compilation works fine.

### Version of Marlin Firmware

Latest Marling 2.1.x bugfix 2023-04-27

### Printer model

Custom

### Electronics

MKS SGEN_L V2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11350190/Configuration_adv.zip)

If I do the same configuration with another motherboard "default_envs = mks_monster8_usb_flash_drive_msc" and BOARD_MKS_MONSTER8_V1, then I DON'T have this problem.

iLLiac4, 2023-04-28:
Hi. Have the same error. Where have you added #define HAS_FAN 1 ?

ellensp, 2023-04-28:
broke on PR https://github.com/MarlinFirmware/Marlin/pull/25332

thinkyhead, 2023-04-28:
This is now patched. Thanks for the report!

closed 2023-04-28 by thinkyhead

rondlh, 2023-04-28:
Confirmed, that fixed it, great job!