MarlinFirmware/Marlin #25753: M119 and home Z (with bltouch) broken

GMagician, 2023-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

It seems that a recent PR broke m119.
In my machine I can home X e Y but not Z (it uses bltocuh) and M119 always reports open for X, Y and Z.

I reverted #25748 and now M119 and home Z both work

### Bug Timeline

Recently

### Expected behavior

Z homes and M119 reports correct state

### Actual behavior

M119 reports always open and Z doesn't home

### Version of Marlin Firmware

bugfix 2.1 up to #25738

### Printer model

Bear 2.1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11350966/Configuration.zip)

thinkyhead, 2023-04-28:
In your troubleshooting, did you find that disabling `ENDSTOP_INTERRUPTS_FEATURE` made any difference?

GMagician, 2023-04-28:
No, I only did a "revert". I'm sorry but I can't do checks in next 4 days.

thisiskeithb, 2023-04-28:
> In my machine I can home X e Y but not Z (it uses bltocuh) and M119 always reports open for X, Y and Z.

Do you home with a BLTouch? `USE_PROBE_FOR_Z_HOMING` is disabled in your config.

ellensp, 2023-04-29:
@thisiskeithb 

USE_PROBE_FOR_Z_HOMING is automatically enabled with  Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
(according to the comments) 

```
/**
 * Enable this option for a probe connected to the Z-MIN pin.
 * The probe replaces the Z-MIN endstop and is used for Z homing.
 * (Automatically enables USE_PROBE_FOR_Z_HOMING.)
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```

thisiskeithb, 2023-04-29:
> USE_PROBE_FOR_Z_HOMING is automatically enabled with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Derp. I knew that. 🤪

GMagician, 2023-05-02:
Fixed in latest bugfix

closed 2023-05-02 by GMagician