MarlinFirmware/Marlin #25755: error compiling

jarvis710, 2023-04-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

problems compiling with fontutils.ccp

### Bug Timeline

i don't if iit's an issue or a bug

### Version of Marlin Firmware

marlin-bugfix-2.1.x

### Printer model

Voxelab aquila x2

### Electronics

btt skr mini3 v3 + btt tft35 v3 + crtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

In file included from Marlin\src\lcd\fontutils.cpp:39:
Marlin\src\lcd\fontutils.h:68:20: error: redefinition of 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)'
   68 | inline const char* get_utf8_value_cb(const char *pstart, read_byte_cb_t cb_read_byte, lchar_t &pval) {
      |                    ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\lcdprint.h:33,
                 from Marlin\src\lcd\marlinui.h:72,
                 from Marlin\src\lcd\fontutils.cpp:35:
Marlin\src\lcd\utf8.h:68:20: note: 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)' previously defined here
   68 | inline const char* get_utf8_value_cb(const char *pstart, read_byte_cb_t cb_read_byte, lchar_t &pval) {
      |                    ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\fontutils.cpp:39:
Marlin\src\lcd\fontutils.h:75:16: error: redefinition of 'uint8_t utf8_strlen(FSTR_P)'
   75 | inline uint8_t utf8_strlen(FSTR_P fstart) { return utf8_strlen_P(FTOP(fstart)); }
      |                ^~~~~~~~~~~
In file included from Marlin\src\lcd\lcdprint.h:33,
                 from Marlin\src\lcd\marlinui.h:72,
                 from Marlin\src\lcd\fontutils.cpp:35:
Marlin\src\lcd\utf8.h:75:16: note: 'uint8_t utf8_strlen(FSTR_P)' previously defined here
   75 | inline uint8_t utf8_strlen(FSTR_P fstart) { return utf8_strlen_P(FTOP(fstart)); }
      |                ^~~~~~~~~~~
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\fontutils.cpp.o] Error 1

thisiskeithb, 2023-04-29:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

This isn't possible since `fontutils.cpp` was renamed in https://github.com/MarlinFirmware/Marlin/commit/9a7d9e6995f4f66c4b48147ec7876d49d4b3ae77 and no longer exists.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

thisiskeithb, 2023-04-29:
You didn't provide your configs, but I cannot reproduce this using the latest `bugfix-2.1.x` (f9db5ab) with a stock Ender-3 with `CR10_STOCKDISPLAY` and an SKR Mini E3 V3 (`BOARD_BTT_SKR_MINI_E3_V3_0`) config.

closed 2023-04-29 by thisiskeithb