MarlinFirmware/Marlin #25763: [BUG] (bug summary)

classicrocker883, 2023-05-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

  in **Configuration_adv.h** 
under `#if HAS_DISPLAY`
having  `#define SHOW_BOOTSCREEN`  and    `#define BOOTSCREEN_TIMEOUT 4000 `
caused issues when booting from *DWIN_LCD_PROUI*
putting back the previous fixed the issue.
```
#if ANY(HAS_MARLINUI_U8GLIB, TOUCH_UI_FTDI_EVE, IS_DWIN_MARLINUI)
...
#endif
```

### Expected behavior

should have booted normally

### Actual behavior

did not go past the first boot screen

### Steps to Reproduce

1. enable a DWIN lcd - Proui

### Version of Marlin Firmware

most current

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

config files to upload not necessary.

ellensp, 2023-05-01:
Config files are always necessary

Should we waste time recreating configs that may or may not trigger this issue, or use a config that is known to trigger the issue and use the time to look into the actual problem?

It also answers a lot of questions. 

For eg, what motherboard are you using? Do you know if it is board specific or not? 

Please attach your configs as requested.

classicrocker883, 2023-05-06:
[Marlinconfigs.zip](https://github.com/MarlinFirmware/Marlin/files/11411754/Marlinconfigs.zip)

my configs may not directly be compatible with Marlin as is, but like I have posted just enable the DWIN ProUI LCD, and this is on a voxelab Aquila board with color 4.3 lcd. it simply would not boot with `SHOW_BOOTSCREEN` or `BOOTSCREEN_TIMEOUT`

classicrocker883, 2023-05-06:
that is a false alarm, not sure what caused the initial non-boot up, but perhaps I mistaken that change in code for something else I did and not recall.

closed 2023-05-06 by classicrocker883