MarlinFirmware/Marlin #25764: [BUG] #define Z_AFTER_HOMING doesn't go to absolute value

sarvenn, 2023-05-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

#define Z_AFTER_HOMING x
It doesn't go to that x value but instead it goes to "X - z_probe_offset_value"

For example. #define Z_AFTER_HOMING 10 , Nozzle to probe offset is -1.5
After homing Z, final Z position is 11.5
It think it should go to absolute position X. Why does it adds the z_probe_offset value into account?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11364295/Configuration.zip)

### Version of Marlin Firmware

Recent Bugfix 2.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-01:
Make sure to use latest bugfix several Z_AFTER_HOMING have been fixed within last 2 weeks

sarvenn, 2023-05-01:
Alright. I will flash a new one today. Also I will update HS mode z clearance to zero. A couple of weeks ago it required to enter a value above 0, and noticed that it is also fixed with a pr.

closed 2023-05-01 by sarvenn