MarlinFirmware/Marlin #25767: [BUG] Compile for ESP32 (MKS Tinybee)

Azrael321, 2023-05-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Current Bugfix is not possible to compile.

`.pio/libdeps/mks_tinybee/ESP3DLib/src/espcom.cpp:25:52: fatal error: ../../../../../Marlin/src/HAL/HAL/ESP32/FlushableHardwareSerial.h: No such file or directory
compilation terminated.
*** [.pio\build\mks_tinybee\lib360\ESP3DLib\espcom.cpp.o] Error 1`

There seems to be one /HAL to much in the file path

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-02:
Issue is with version of ESP3DLib used
Path was changed in Marlin  https://github.com/MarlinFirmware/Marlin/pull/25743

so the library also needed updating. https://github.com/luc-github/ESP3DLib/pull/68

check Marlin/.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h
if it says #define LIB_VERSION "1.0.11"   this is not the latest version of the library should be 1.0.12

delete the directory .cache in .platformio under you user directory
delete .pio directory under Marlin
 
rebuild, this forces platformio to get the latest version.

closed 2023-05-02 by ellensp

ellensp, 2023-05-02:
Here is a minimal config that can trigger issue.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11371327/Configuration.zip)

santocyber, 2023-05-20:
here the same issue more or less the same on marlin 2.1.2.1 but here the error is

In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/command.cpp:21:0:
.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: /home/luiza/Documents/AMYWORK3D/AIMPRESSORA-GRANA3D/MINI/inc/MarlinConfigPre.h: No such file or directory
compilation terminated.
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlib.cpp:27:0:
.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: /home/luiza/Documents/AMYWORK3D/AIMPRESSORA-GRANA3D/MINI/inc/MarlinConfigPre.h: No such file or directory
compilation terminated.
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.cpp:21:0:
.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: /home/luiza/Documents/AMYWORK3D/AIMPRESSORA-GRANA3D/MINI/inc/MarlinConfigPre.h: No such file or directory
compilation terminated.
*** [.pio/build/mks_tinybee/libcf4/ESP3DLib/command.cpp.o] Error 1
*** [.pio/build/mks_tinybee/libcf4/ESP3DLib/esp3dlib.cpp.o] Error 1
*** [.pio/build/mks_tinybee/libcf4/ESP3DLib/esp3dlibconfig.cpp.o] Error 1

santocyber, 2023-05-20:
> Issue is with version of ESP3DLib used Path was changed in Marlin #25743 so the library also needed updating. [luc-github/ESP3DLib#68](https://github.com/luc-github/ESP3DLib/pull/68) Both of these have been merged. (ie completed a week ago)
> 
> check Marlin/.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h if it says #define LIB_VERSION "1.0.11" this is not the latest version of the library should be 1.0.12
> 
> delete the directory .cache in .platformio under your user directory delete .pio directory under Marlin
> 
> rebuild, this forces platformio to get the latest version.

i verify my esp3dlibconfig and library is updated

#include MARLIN_PATH(inc/MarlinConfigPre.h)
#undef DISABLED
#undef _BV
//version
#define LIB_VERSION "1.0.12"

santocyber, 2023-05-20:
> here the same issue more or less the same on marlin 2.1.2.1 but here the error is
>
> In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/command.cpp:21:0: [...]

i found the issue, is folder have a space, when rename the folder without space build is fine thanks

SeamusCheng, 2023-06-01:
I also encountered the same problem. May I ask which folder you mentioned has spaces?

> 

I also encountered the same problem. May I ask which folder you mentioned has spaces?

SeamusCheng, 2023-06-01:
> i found the issue, is folder have a space, when rename the folder without space build is fine thanks

I also encountered the same problem. May I ask which folder you mentioned has spaces?

PanosPetrou, 2023-06-01:
Make sure your path to the Marlin files (e.g. "C:\Firmware Files\Marlin-Bugfix") doesn't have a folder name that contain spaces. For example in "C:\Firmware Files\Marlin-Bugfix", the folder "Firmware Files" must be renamed because it contains a space.

SeamusCheng, 2023-06-01:
I still have the same problem moving it to the root directory of the D drive

PanosPetrou, 2023-06-02:
Redownload Marlin and start over

SeamusCheng, 2023-06-02:
.pio/libdeps/esp32/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h: No such file or directory  
compilation terminated.
In file included from .pio/libdeps/esp32/ESP3DLib/src/espcom.cpp:21:0:
.pio/libdeps/esp32/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h: No such file or directory  
compilation terminated.
In file included from .pio/libdeps/esp32/ESP3DLib/src/esp3dlib.cpp:27:0:
.pio/libdeps/esp32/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h: No such file or directory  
compilation terminated.
*** [.pio\build\esp32\libe78\ESP3DLib\command.cpp.o] Error 1
In file included from .pio/libdeps/esp32/ESP3DLib/src/esp3dlibconfig.cpp:21:0:
.pio/libdeps/esp32/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h: No such file or directory  
compilation terminated.
*** [.pio\build\esp32\libe78\ESP3DLib\esp3dlib.cpp.o] Error 1
*** [.pio\build\esp32\libe78\ESP3DLib\espcom.cpp.o] Error 1
*** [.pio\build\esp32\libe78\ESP3DLib\esp3dlibconfig.cpp.o] Error 1
=============================================================== [FAILED] Took 78.95 seconds ====Re download and recompile the same issue

SeamusCheng, 2023-06-02:
[image]

PanosPetrou, 2023-06-02:
Your path "C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h" is missing backslashes, it doesn't seem correct.
Try the following: 
Go to C:\ and create a new folder and name it Firmware. 
Download the latest Marlin bugfix from https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip 
Extract everything into that folder and retry.

thisiskeithb, 2023-06-02:
Also:

1. Delete the hidden `.pio` folder at the root of your Marlin project.

2. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

3. Re-install PlatformIO.

This will force all new downloads of libraries / toolchains & reset PlatformIO in case anything is broken there.

SeamusCheng, 2023-06-03:
I found that Marlin is very unfriendly to the motherboard of ESP32, which is much more powerful than the mega2560 and cannot enable EMERGENCY _ PARSER, I can't use commands such as M118 on my serial screen anymore. Forget it, let's switch to a Mega2560 motherboard instead. I don't want to mess around anymore.

okelk, 2023-06-29:
> Your path "C:Users12077DesktopMarlinMarlin/src/inc/MarlinConfigPre.h" is missing backslashes, it doesn't seem correct. Try the following: Go to C:\ and create a new folder and name it Firmware. [...]

I seem to have the same problem using VSCODE / platformio under windows 10
no idea if this problem is in this code or with platformio
every time MARLIN_PATH is expaded it looses all the backslashes.
Any idea why this might be? -maybe it works  just fine under operating systems that use normal slashes? /

xkub61, 2023-07-31:
The solution i figured out for this problem with the messed up path was just compile the firmware on linux, and it worked. Just make a dual boot on your pc and it will work fine.

booroondook-eng, 2023-09-25:
The following workaround really helped me:
1. Go to your project's **.pio\libdeps\mks_tinybee\ESP3DLib\src** directory.
2. Find the **esp3dlibconfig.h** file and open it for editing.
3. Find the following lines (maybe #31 and #32):
```
#define MARLIN_HAL_PATH(PATH) HAL_PATH(HALHOME, PATH)
#define MARLIN_PATH(PATH) ESP_XSTR(SRCHOME/PATH)

```
4. Replace these two lines with another two lines:
```
#define MARLIN_HAL_PATH(PATH) HAL_PATH( ../../../../../Marlin/src/HAL, PATH)
#define MARLIN_PATH(PATH) ESP_XSTR(../../../../../Marlin/src/PATH)
```
5. Save the file and try to compile your project.