MarlinFirmware/Marlin #25771: [FR] Add Heated bed heating as a condition to the PSU AUTO_POWER_CONTROL

ovcara, 2023-05-02:
### Is your feature request related to a problem? Please describe.

I'm using SKR v3 EZ and I'm powering it with ATX PSU 5V "Stand By" to supply board's logic with the power. PS_ON pins have ATX line and GND attached. On power-up PSU's main switch, printer starts up normally in "Stand By mode". Using LCD controller, I switch on the PSU in order to power the board with 12V. As soon as I set the temperature to the heated bed, the board shuts down the PSU, leaving only 5V stand by to power the printer, which is not enough to bring up heated bed to the desired temperature.

By examining the "Configuration.h" I noted that Marlin 2.1.2 in "AUTO_POWER_CONTROL" checks only, if fans need power or, if printers hotend is too hot. Only then it will shut down or power up.

### Are you looking for hardware support?

Using a printer of my own make with SKR 3 EZ - don't need support for the hardware.

### Describe the feature you want

I'm recommending to add additional condition to check, if some other feature (i.e. heated bed) requires power, and to keep PSU ON or powered, as usually printers are cold and don't need fans to run initially. This feature would contribute also to the PSU_DEFAULT_OFF and after the temperature of the heated bed is set (or start of print) PSU would be switched on.

Graylag-PD, 2023-05-03:
I am running 2.1.2 and I also have printer configured in a way, that it starts with PSU switched off, with switching on only when needed (using AUTO_POWER_CONTROL) and I can confirm that the feature that you ask is already implemented. When you set target temperature to either hotend or heatbed, PSU automatically switches on. No need to manually switch PSU on. 
BUT!
There is one strange anomaly. When you set your hotend temperature, PSU switches on immediately. But when you set your heatbed temperature PSU switches on only after you return to status screen (does not switch on if you stay in menu). This actually goes so far, that if you set only heatbed temperature and never return to status screen, printer will eventualy activate thermal runaway shutdown because it thinks heatbed thermistor decoupled. 
Try it yourself if it behaves same way as mine does and if you want, you can even try to write a bug report on it.

ovcara, 2023-05-03:
Thank you for your response. I tried to follow you instructions but failed. This is what I did:
1. With 5Vsb line connected to the board, I switched on the main power switch on the PSU.
2. The board is booting in normal mode, with PSU staying switched off (normal behaviour as expected).
3. Setting the Bed temp to 40°C, and returning back to the status screen did not switch on the PSU, but was followed by a thermal runaway after some time. (test stopped)
Second test:
Steps 1. and 2. as previously.
3. Setting the Hotend temperature to 61°C, and in the same instance as I confirmed the set temp, PSU went on and heated to the temp.
4. Having hotend hot, I set the Heatbed temperature to 40°C and as I confirmed it, the PSU went off. Returning to the Status screen, does not power it on again.

The only way these features work is that I set hotend to desired temperature, disconnect 5Vsb from the board and just after that start heating the heatbed. 

At this point I don't really know is this a bug in Marlin, is something wrong with my board or Marlin is missing some feature?

I know it did work in Marlin 2.1.1, but at that time I was running it on RAMPS v1.4. Recently I upgraded my printer to BTT SKR V3 EZ.

GMagician, 2023-05-03:
It seems that bed is considered for poweron.
`if (TERN0(HAS_HEATED_BED, thermalManager.degTargetBed() > 0 || thermalManager.temp_bed.soft_pwm_amount > 0)) return true;`

GMagician, 2023-05-03:
But I can't test it, no PSU with PS_ON

ovcara, 2023-05-04:
> It seems that bed is considered for poweron. `if (TERN0(HAS_HEATED_BED, thermalManager.degTargetBed() > 0 || thermalManager.temp_bed.soft_pwm_amount > 0)) return true;`
> 
> Edit: but printer is considered in pause when `ui.on_status_screen` that may justify @Graylag-PD, but this should not be different for bed and hotend

While this piece of code sounds logical to me (even if I'm no programmer), I don't know where to put it. In Marlin/source/feature/power.cpp maybe? Where exactly?
I could run some tests then.

GMagician, 2023-05-04:
This code is already present in Marlin and should do what you are asking, can't understand why it doesn't.
Only thing I see is 'pause'. What does happens if you comment out POWER_TIMEOUT?

ovcara, 2023-05-04:
It does exactly the same thing as if I leave it uncommented. No change.
Another two things I noticed just now are:
If I select to print something from SD card, just after selection and confirmation, the printer goes back to status screen, with system message "Bed Heating..." and powers on for about half a second and then PSU goes back off. If I scroll down on the menu at that time there is function "Switch Power Off". It seems that printer "thinks" that the power is on, but it is not. Something just prevents it from normal functioning.

The second thing I did was that I selected at first to heat up the Nozzle. The PSU goes instantly on and heating starts. If just after that I select to heat up the bed, the PSU goes down in the same moment with all above symptoms. I'm actually worried because nevertheless power is off, the nozzle is still gaining temperature. Which means, the board is using power coming out of 5V Stand By line.

Currently the only way I can print and use the printer is to Power it on (via menu), disconnect the 5Vsb and then select what I want to print. - That works, but... If I connect the 5Vsb line during printing process, everything stops immediately and the PSU goes off.

That all made me thinking, that maybe it is not the firmware issue but board issue. Or maybe I'm not using the correct PIN to provide the board with 5Vsb. I'm not sure anymore.

Graylag-PD, 2023-05-05:
just an idea, but could it be that your PSU shuts itself down because of overload? What is its rating and what is the heatbed power? Do you know how that specific PSU handles overload? 
In my case I have separate PSU just for bed and it is in fact underpowered, but it handles it by limiting current without shuting down. 

Or maybe the power from 5V supply leaks to main PSU and causes a havoc within its circuitry. I have BTT Octopus Pro and I do know for a fact, that power from 5V auxiliary source leaks into main 24V supply when that one is shut down. Mine PSU does not mind, but yours might. 

Or maybe it is a combination of both, that supply gets overloaded and it would have handled it somehow, but that 5V leak confuses itself into full shutdown. 

This does not change anything on the fact, that AUTO_POWER_CONTROL could use a fix in relation to heatbed, but it might not be what causes your issues.

ovcara, 2023-05-07:
I'm running my printer on regular PC converted ATX PSU, (HKC USP-5550) with 550 W of power, capable of delivering 24Amps on 12V, 2.5Amps on 5Vsb, etc. (you can see more on https://manualsbrain.com/en/manuals/139606/?page=6). I have no idea how it handles overloads as I don't know its circuitry, but I'm suspecting there shouldn't be any. The heatbed power rating is 120W.
I made some measurements of Voltage drops in its idle state (but on) and loaded. The PSU gives out 12.6 Volts unloaded and 11.84 Volts when heat bed heating is on. The 5Vsb is at 5.05 V at all times.

I'll just make some workaround this issue and install additional normally closed relay on the 5Vsb, so it shuts down 5Vsb when PSU Powers on and increase the delay to about 1000 ms in Marlin. This should enable proper functioning also vice versa, when the PSU shuts down, as its capacitors still retain some power for about 4 to 5 seconds.

Graylag-PD, 2023-05-07:
Droping to 11,84 at just 10A output (42% of limit) looks a bit suspicious to me. It is within ATX specification, so it might not be an issue, but it looks suspicious. If you can, I would suggest testing it with different PSU. 
ATX supplies should in general shutdown when overloaded. 

By the way nominal rating of bed can be very different to its actual power. Try to measure resistance of the bed in cold state. Keep in mind, that power of the board in cold state can be up to 50% higher than in warm state, which can make huge difference to PSU load. 

But those are just suggestions. I don't understand how does disconnection of 5VSB makes difference, nor I can be sure if it is purely HW issue, or also SW issue. I did use ATX supply to power my printer for few years in similar configuration, but I never observed this kind of behaviour.

ovcara, 2023-05-18:
Sorry not to respond earlier.
I can't do the test with another ATX PSU - I don't have additional one. I threw away 2 even older ones (one 300W and one 350W) than this one I'm currently using.

I measured the resistance of old and a new bed and both of them have 1.2 Ohms in cold state. Tried with both of them and results are the same. (BTW, I bought new heated bed purely because of its dimensions - 2 cm larger than the old one - not because I would suspect something is wrong with the smaller one).

closed 2023-07-29 by github-actions[bot]