MarlinFirmware/Marlin #25775: [BUG] Unable to revert from bugfix-2.1.x to 2.1.2; SKR Mini E3 won't boot

jantman, 2023-05-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When reporting #25774, as requested in the issue template, I compiled the current bugfix-2.1.x ( as of e1f6435d4410478930ef5fb7d99a311332f3cee8 ) and flashed that to ensure that the bug still exists on that branch. I then tried to re-flash back to the same binary for 2.1.2, and the board will not boot. I get a blinking activity LED for the SD card, and the `firmware.bin` file is removed, but the board will not boot - nothing on the display, and nothing out on USB. This is using the same bin that was successfully running earlier today. If I re-flash with my build of bugfix-2.1.x, it flashes and boots fine. If I again attempt to flash back to one of my 2.1.2 builds from earlier today, it again will not boot.

### Bug Timeline

Today.

### Expected behavior

Expected my board to boot to the 2.1.2 firmware.

### Actual behavior

Board does not boot - nothing on display, nothing on USB, no outward signs of life other than the power LED.

### Steps to Reproduce

1. Flash board to working 2.1.2 build. Everything works fine.
2. Flash board to bugfix-2.1.x build. Everything works fine.
3. Attempt to flash back to same binary from Step 1 (2.1.2), and board does not boot.
4. Flash board back to bugfix-2.1.x build. Everything works fine.
Repeat seemingly indefinitely.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality CR-10S

### Electronics

BTT SKR Mini E3 V3.0

### Add-ons

PT100 2-wire RTD and MAX31865 board (Adafruit 3328)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

* Original, working 2.1.2 firmware:
   * Configuration: https://github.com/jantman/3d-printed-things/tree/71fedd2ea95d96765beeb07b31b1e1b0934c4248/my-cr10s/marlin-configs
   * Binary and config.ini: https://github.com/jantman/3d-printed-things/releases/tag/build34_71fedd2
   * Build output: https://github.com/jantman/3d-printed-things/actions/runs/4866133345/jobs/8677317214
* bugfix-2.1.x that can't be downgraded:
   * Configuration: https://github.com/jantman/3d-printed-things/tree/a3b5b02b597f391ace511bc531d432827974cf6d/my-cr10s/marlin-configs
   * Build output: https://github.com/jantman/3d-printed-things/actions/runs/4866635218/jobs/8678369730
   * Binary and config.ini (ZIP file): https://github.com/jantman/3d-printed-things/suites/12632715201/artifacts/676503398

PendulumPit, 2023-05-03:
Leave the board on for 4 to 5 mins and it will start with your 2.1.2 flash. I went through the same thing on my octopus board and SKR2 board. Another person same thing with their 4.2.7.  I tried this a few times and it repeats itself. I was told it was nothing to do with Marlin itself.  Strange things happen I guess

EvilGremlin, 2023-05-03:
Yep, binary content have nothing to do with ability to flash. Boot will happily flash anything at all, as long as filename and size is right.

closed 2023-05-03 by thisiskeithb

jantman, 2023-05-03:
Thanks so much, @PendulumPit . Confirmed working.

Note, I'm not sure if it's related (and have zero desire to try and reproduce at this point), but I waited 8 minutes after flashing finished, and was still stuck at a blank screen. I then removed the SD card, and it booted within ~2 seconds.

Yeah, I'm happy to leave this one at "strange things happen" :)