MarlinFirmware/Marlin #25776: SanityCheck.h:2044:8: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS." [BUG] (bug summary)

Elchschmied, 2023-05-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Today I can't compile my Marlin version for my printer with MMU2S
SanityCheck.h:2044:8: error: #error "TEMP_SENSOR_1 is required with 2 or more HOTENDS."
MMU2S uses only one hotend but extruder count is 5

[image]

### Bug Timeline

new error

### Expected behavior

I expect to fix issue for MMU2S and single hotend

### Actual behavior

It does not compile, see screenshot

### Steps to Reproduce

1. open configuration, board is SKR 1.4 Bigtreetech
2. start compiling
3. system throws sanity check errors

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02010300 from bugfix of 20230503

### Printer model

Easyframe diy

### Electronics

Bigtreetech SKR 1.4 LPC1768

### Add-ons

MMU2S

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Easyframe_config.zip](https://github.com/MarlinFirmware/Marlin/files/11379779/Easyframe_config.zip)
Version pulled this morning from following source
https://github.com/MarlinFirmware/Marlin/commit/dbed3f1c418e1f6edc4d74f5e29659245f6841b0

ellensp, 2023-05-03:
#define SINGLENOZZLE

closed 2023-05-03 by ellensp

Elchschmied, 2023-05-03:
No, next problem then occurs. 
[image]

ellensp, 2023-05-03:
you don't even have a MMU defined

#define MMU_MODEL PRUSA_MMU2

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Elchschmied, 2023-05-03:
Upps, it was defined before and I already use MMU2S. Don't know why it was commented now. I only made a merge of the new bugfix this morning.

ellensp, 2023-05-03:
https://github.com/MarlinFirmware/Marlin/commit/a872232b15d63c73c5c5ce281e3d1dcadb47b608

You will have merged this without updating it

Elchschmied, 2023-05-03:
May be that I missed it. I did it often before with no issue. Problem mostely 50cm infront of the screen.