MarlinFirmware/Marlin #25782: [BUG] MKS_ROBIN2 env missing

texscalper, 2023-05-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using a MKS_ROBIN2 BOARD , latest Marlin 2.1.x dose not compile ... 

PS E:\3D\Arduino\Predator\MKS Robin 2\MKS-Robin2-master\firmware\Marlin-bugfix-2.1.x_Robin2> platformio run --silent -e MKS_ROBIN2
Error: Unknown environment names 'MKS_ROBIN2'. Valid names are 'STM32F103RC_creality_xfer, ........

### Expected behavior

In file stm32f4.ini there is no env for MKS_ROBIN2 (STM32F407ZET6) 

### Actual behavior

Error compiling

### Steps to Reproduce

1. Configuration.h setup MKS_ROBIN2 board  
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_ROBIN2
#endif

2. compile Marlin with env 
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = MKS_ROBIN2
include_dir  = Marlin

3. error...compiling

### Version of Marlin Firmware

2.1.x

### Printer model

bulding it

### Electronics

upgrading

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11405157/Configurations.zip)

ellensp, 2023-05-05:
so it is...

Here is the original 
https://github.com/makerbase-mks/MKS-Robin2/blob/master/firmware/Marlin2.0-MKS-Robin2/platformio.ini#L421-L433

thisiskeithb, 2023-05-05:
We had it at one point: https://github.com/MarlinFirmware/Marlin/pull/16270

ellensp, 2023-05-05:
The PR just added the board pins, not the env  or variants files....

closed 2023-05-08 by thinkyhead