MarlinFirmware/Marlin #25785: [BUG] Latest updates (2023-05-05) cause changed movement direction issues (Z2 and X)

rondlh, 2023-05-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After the updates of 2023-05-05 I see different movement direction for Z2 and my X axis doesn't reverse direction anymore.
There might be more issues, but I gave up when I saw this.
It's an IDEX system with i3 design.

### Bug Timeline

2023-05-05, 2023-05-03 worked as expected

### Expected behavior

Behave like before, correct/unchanged axis movement direction

### Actual behavior

Z2 is reversed, X can only move in 1 direction

### Steps to Reproduce

Move the axis with G-code, G0 X10, G0 Z1

### Version of Marlin Firmware

Lateset bugfix 2023-05-05

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BL Touch V3.1, BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11409372/Configuration_adv.zip)

RV-from-be, 2023-05-05:
@thinkyhead 
I can only confirm this disastrous bug that led to me to the destruction of my belt in X!
The Commit in question is https://github.com/marlinfirmware/marlin/commit/2538a7c4d6148d8249f5418a76c58ec483343D2B. He takes up the famous PR where we have already told you about major problems of axes control.
On an Ender3 V427, the X and Y axis rides must on the endstops without populations arrest them, stop or reversed the directions. I came back before this commit and everything has become normal again!

thisiskeithb, 2023-05-05:
Looks like some changes were reverted: https://github.com/MarlinFirmware/Marlin/commit/8425ab02c954427f509ab5be54c52d573a8995fc

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

thinkyhead, 2023-05-05:
Sorry about that bug. I pushed the changes from the wrong branch, which still had a union bug in the `AxisBits` type. I have since pushed the corrected `AxisBits` from my test setup where the union was fixed, after going through similar axis grinding fun … but thankfully no broken belts.

rondlh, 2023-05-06:
> Sorry about that bug. I pushed the changes from the wrong branch, which still had a union bug in the `AxisBits` type. I have since pushed the corrected `AxisBits` from my test setup where the union was fixed, after going through similar axis grinding fun … but thankfully no broken belts.

I tested the code after the AxisBits update was undone. X movement now works, but my Z2 is still reversed. 
I added:
#define INVERT_Z2_DIR true
Which solved the issue. I previously only had "#define INVERT_Z_DIR  true".

Not sure what happened, but if now undefined INVERT_2Z_DIR defaults to false (I think it should default to INVERT_Z_DIR), then this will cause lots of bend lead screws.
!!! BE CAREFUL, TEST Z-AXIS WITH 0.1MM STEPS FIRST!!!

thisiskeithb, 2023-05-06:
> X movement now works, but my Z2 is still reversed.

Behavior was a bit different for me when I flashed the latest `bugfix-2.1.x` (4528fde).

My test machine is an SKR 3-powered Ender-3 with with dual, independent Z drivers/motors. EEPROM is reset automatically after every flash.

After the initial flash, X & Y moved in the correct direction, but all Z movements were reversed (at the same time, so one motor wasn't reversed compared to the other like rondlh).

Despite that, I tried homing Z, which moved in the correct direction (toward the bed) and all further movements for each axis were correct.

I rebooted the board and without homing and all movements were correct.

Very...strange.

thisiskeithb, 2023-05-06:
> X movement now works, but my Z2 is still reversed.

Behavior was a bit different for me when I flashed the latest `bugfix-2.1.x` (4528fde).

### Ender-3 with an SKR V3
Independent, dual-Z drivers & motors

After the initial flash, X & Y moved in the correct direction, but all Z movements were reversed (at the same time, so one motor wasn't reversed compared to the other like rondlh).

### Biqu B1 SE Plus with SKR V2
Independent, dual-Z drivers & motors

After the initial flash, X & Y moved in the correct direction, but all Z movements were reversed (at the same time, so one motor wasn't reversed compared to the other like rondlh).

### Prusa MK3S+ Bear with a BTT002
Dual-Z motors connected to a single driver

After the initial flash, X & Y moved in the correct direction, but all Z movements were reversed (at the same time, so one motor wasn't reversed compared to the other like rondlh).

In all cases, Z movement is reversed until you home.

thisiskeithb, 2023-05-06:
I've been chatting with thinkyhead and he just pushed [🐛 Apply direction bits at startup](https://github.com/MarlinFirmware/Marlin/commit/d93aa0c38449bcbc38aeca7eac506297270546c8) / d93aa0c which solves my particular Z movement issues, so maybe they'll help?

Theo9909, 2023-05-06:
SKR3 Dual Z motors on separate drivers. After initial power on, the Z-axis steppers go in opposite direction. After initial move, they then move in sync. I'll pull the pushes to my git and see what happens. Finger crossed.

thinkyhead, 2023-05-07:
I found two issues that could affect the DIR state and patched them both. If everything is solved, you can go ahead and close this issue.

_Mea culpa!_ Going forward I'll be doing more testing of the upcoming motion and endstop patches and will be sure to refrain from pushing "simple cosmetic updates" when I'm lacking sleep. 💤

rondlh, 2023-05-07:
> I found two issues that could affect the DIR state and patched them both. If everything is solved, you can go ahead and close this issue.
>
> _Mea culpa!_ Going forward I'll be doing more testing of [...]

No worries, Marlin has become a very powerful, flexible and complex firmware, issues are bound to happen, that's why there is a bugfix branch. Your efforts are very appreciated!

thisiskeithb, 2023-05-07:
@rondlh: Can this issue be closed?

Theo9909, 2023-05-07:
Tested on my machine and it works as it should. No more issues with Z right after the machine is first powered on. X and Y function as they should. I second closing it.

closed 2023-05-07 by thinkyhead

rondlh, 2023-05-08:
> @rondlh: Can this issue be closed?

Yes!