MarlinFirmware/Marlin #25788: [BUG] Parameter values are not saved in EEPROM

trengtor, 2023-05-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After I tried using 2.1.2 and 2.1.bugfix, I had to revert to 2.0.8.2. Reason for return: did not like it, so far everything is rather unstable.

After returning to 2.0.8.2, I found a problem: after a reboot, the values previously saved in EEPROM disappear. In particular, the HYBRID THRESHOLD parameters. Neither the initialization of the EEPROM, nor the loading of default values, nor any other hits to the collapsible shaman's tambourine helps.

As far as I understand, in the 2.1 branch, something has changed a lot in working with the EEPROM and now something prevents it from initializing and saving parameters, if we take the firmware of the 2.0 branch.

I don't like to re-initialize the EEPROM every time I print. Therefore I ask a question: how to fix it?

Dear developer team! I look forward to your suggestions on how to solve this problem.

### Version of Marlin Firmware

2.0.8.2

### Printer model

selfmade

### Electronics

BTT E3 RRF V1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

trengtor, 2023-05-07:
Will there be any recommendations on how to revive the work of EEPROM after a downgrade from 2.1.2 / 2.1.bugfix to 2.0.8.2? Or is the salvation of the drowning the work of the drowning themselves? And the developers of Marlin, as it were, have nothing to do with it?

ellensp, 2023-05-07:
2.0.8.2 was released May 30, 2021, almost two years old at this stage.
no one is working on old code.

thisiskeithb, 2023-05-07:
You likely caught `bugfix-2.1.x` in a bad state, but it's updated nearly every day and is back to being stable for most users as of (d36f43d).

> Will there be any recommendations on how to revive the work of EEPROM after a downgrade from 2.1.2 / 2.1.bugfix to 2.0.8.2? 

As ellensp stated, we won't be providing support for a version of Marlin that old and any potential EEPROM-related bugs would have likely been fixed since then.

For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-05-07 by thisiskeithb

trengtor, 2023-05-07:
To be honest, I had little doubt that your answer would be exactly that, guys.

trengtor, 2023-05-07:
May be anubody will help:

[image]
[image]