MarlinFirmware/Marlin #25789: Problem with building with "auto build marlin"

Ale3xDARS, 2023-05-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When clicking build the following commands pop up and every time the same error comes up.

No matter the version of marlin or cofig folder.

PS C:\Users\Fucking Good\Documents\Arduino\_Marlin\Marlin> platformio run --silent -e STM32F103RE_creality
Error: Could not find one of 'package.json' manifest files in the package
PS C:\Users\Fucking Good\Documents\Arduino\_Marlin\Marlin> echo "done" >"C:\Users\FUCKIN~1\AppData\Local\Temp\ipc"

PS: Yes my Desktop name is "Fucking Good".

### Steps to Reproduce

Following steps from various videos on how to upload the marlin firmware onto the Creality Silent Motherboard V4.2.7 this same error appears

### Version of Marlin Firmware

MArlin 2.0.X

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-06:
Issue with platfromio, not marlin,

delete .platformio directory in you home directory and restart vscode. give it time to reinstall. it has to get a lots of data from the internet.

Also put marlin somewhere else. platformio doesn't like spaces in the path

closed 2023-05-06 by ellensp

Ale3xDARS, 2023-05-06:
It appears I do not understand what you are trying to say...

Ale3xDARS, 2023-05-06:
> Issue with platfromio, not marlin,
> 
> delete .platformio directory in you home directory and restart vscode. give it time to reinstall. it has to get a lots of data from the internet.
> 
> Also put marlin somewhere else. platformio doesn't like spaces in the path

It appears I do not understand what you are trying to say...

Ale3xDARS, 2023-05-06:
As I am new to all this, would it be OK with anyone willing to help me via discord or any other way??