MarlinFirmware/Marlin #25794: Dual Z motors, the REV setting only works on one side

colcolcheese, 2023-05-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

#define INVERT_Z_DIR false => Z&Z2=right rotation
#define INVERT_Z_DIR true => Z=left  rotation Z2=right rotation

### Version of Marlin Firmware

bugfix-2.1.x 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11421699/Marlin.zip)

ellensp, 2023-05-08:
actually  not so sure...   leaving open

ellensp, 2023-05-09:
git bisect says commit d93aa0c38449bcbc38aeca7eac506297270546c8 introduced this issue.

thisiskeithb, 2023-05-09:
> says commit [d93aa0c](https://github.com/MarlinFirmware/Marlin/commit/d93aa0c38449bcbc38aeca7eac506297270546c8) introduced this issue.

That patch was pushed to fix the motion bug(s) reported in https://github.com/MarlinFirmware/Marlin/issues/25785

ellensp, 2023-05-09:
Looking into the code, and getting rather confused, but I think I see the issue

The code is correct, but ...

what is not happening is when
#define INVERT_Z_DIR true 
then 
 #define INVERT_Z2_DIR true 
should also be set. but it is not
simply adding the line will fix the issue, but it should be done automatically 

I purpose the following, which fixes it in my testing 

```DIFF
diff --git a/Marlin/src/inc/Conditionals_adv.h b/Marlin/src/inc/Conditionals_adv.h
index 7ea12f0493..c0357ba78d 100644
--- a/Marlin/src/inc/Conditionals_adv.h
+++ b/Marlin/src/inc/Conditionals_adv.h
@@ -875,6 +875,19 @@
 #endif
 
 // Multiple Z steppers
+#ifdef INVERT_Z_DIR
+  #if NUM_Z_STEPPERS > 3
+    #define INVERT_Z4_DIR INVERT_Z_DIR
+  #endif
+  #if NUM_Z_STEPPERS > 2
+    #define INVERT_Z3_DIR INVERT_Z_DIR
+  #endif
+  #if NUM_Z_STEPPERS > 1
+    #define INVERT_Z2_DIR INVERT_Z_DIR
+  #endif
+#endif
+
+#define INVERT_Z4_DIR INVERT_Z_DIR
 #if NUM_Z_STEPPERS < 4
   #undef INVERT_Z4_VS_Z_DIR
   #if NUM_Z_STEPPERS < 3

```

colcolcheese, 2023-05-09:
The test was successful.
Thanks for responding.

ellensp, 2023-05-09:
PR created, closing this as fixed. (as soon as its is committed)

closed 2023-05-09 by ellensp