MarlinFirmware/Marlin #25797: G1 with E hangs after disabling CLASSIC_JERK

paulorlyk, 2023-05-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After disabling CLASSIC_JERK the following code hangs indefinitely on the last line.

```M201 X2000 Y2000 Z20 E3000
M203 X1000 Y1000 Z12 E60
M204 P3000 R3000 T3000
M205 X8.00 Y8.00 Z0.20 E10.00
M205 S0 T0

G90

M140 S60
M104 S205
M190 S60
M109 S205

M420 S1
G28 ; home all axis

G0 Z3
M201 X50000 Y50000
M204 P1000 T1000
M205 J0.02

T0
G92 E0.0000
G1 E-0.7 F3000

G1 Z0.200 F1200
G1 X109.9080 Y187.7420 F6000
G1 E0 F3000
G92 E0.0000
G1 X109.9080 Y106.4240 E3.5052 F1800
```

The ```G1 X109.9080 Y106.4240 E3.5052 F1800``` causes printer to make a very short jerky movement and hang indefinitely with repeating "busy: processing" messages sent to host.

When I remove E parameter, ```G1 X109.9080 Y106.4240 F1800``` executes fine.

I think (but not sure) that only ```G1 X109.9080 Y106.4240 E3.5052 F1800``` by itself is enough to cause the issue and the rest of G-code can be omitted.

I have LIN_ADVANCE enabled.

### Expected behavior

```G1 X109.9080 Y106.4240 E3.5052 F1800``` does not take forever to execute

### Actual behavior

```G1 X109.9080 Y106.4240 E3.5052 F1800``` hangs with "busy: processing"

### Steps to Reproduce

Run ```G1 X109.9080 Y106.4240 E3.5052 F1800```

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (May  9 2023 02:36:34)

### Printer model

Cusom CoreXY

### Electronics

BOARD_BTT_SKR_V1_4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Maybe related to [25553](https://github.com/MarlinFirmware/Marlin/issues/25553) [25566](https://github.com/MarlinFirmware/Marlin/issues/25566)

My configuration: [config.zip](https://github.com/MarlinFirmware/Marlin/files/11429616/config.zip)

thinkyhead, 2023-05-30:
We should look closer at this one @tombrazier …

tombrazier, 2023-05-30:
Well, being repeatable is an excellent start. Excellent work narrowing it down @paulorlyk.

I'll try to have a look. Hopefully it also happens on a processor I have.

paulorlyk, 2023-05-31:
It's been a while, so I decided to test against the latest bugfix-2.1.x. It did not reproduce. So I reverted back to commit f80e706f. It did not reproduce either!

After many trials and errors, I believe, I found the issue:

It appears that I am incorrectly building the firmware:
```
$ pio run --target clean -e LPC1768
$ pio run -e LPC1768
```

If, instead of running `$ pio run --target clean -e LPC1768`, I remove `.pio` directory, everything works fine.

After running `$ pio run --target clean -e LPC1768`, `.pio/build` contains only `project.checksum` file. I was not able to find any differences in `.pio/libdeps` between working and broken builds...

paulorlyk, 2023-05-31:
(Deleted my previous comment because it was totally incorrect and misleading)

It seems not reproducible after [1f9bfc5c](https://github.com/MarlinFirmware/Marlin/commit/1f9bfc5c7491d802d242623f8253b1f52c0917a3) 🚸 Optional X-Axis (#25418)

thisiskeithb, 2023-05-31:
> It seems not reproducible after [1f9bfc5c](https://github.com/MarlinFirmware/Marlin/commit/1f9bfc5c7491d802d242623f8253b1f52c0917a3) 🚸 Optional X-Axis (https://github.com/MarlinFirmware/Marlin/pull/25418)

Glad it’s fixed now 👍

closed 2023-05-31 by thisiskeithb

paulorlyk, 2023-06-23:
I am on [60f591f3 [cron] Bump distribution date (2023-06-23)
](https://github.com/MarlinFirmware/Marlin/commit/60f591f3b6e06fdef903e10f81ac96f65e30d78e) and the issue is back. Not sure at this point if it's some sort of memory corruption type of issue, or I am doing something wrong. I apologize if the later is the case.