MarlinFirmware/Marlin #25801: [FR] Firmware for Bambulabs X1-Carbon?

Kreyren, 2023-05-09:
### Is your feature request related to a problem? Please describe.

I am working on https://github.com/Ultimaker/Cura/pull/15474 and it's paiinnn because of the non-standard firmware used

### Are you looking for hardware support?

Bambulabs X1-Carbon

### Describe the feature you want

Alternative open-source firmware

### Additional context

My motivation is being able to use cura with activism for liberating the machine.

thinkyhead, 2023-05-30:
What do you know about the mainboard of the Bambu? It has some custom components such as the LIDAR camera that we would not be likely to be able to support, and the accelerometer would also require entirely new code. The LCD controller is also entirely new to us. So it's going to be a kong road to supporting the Bambu.

Kreyren, 2023-06-01:
> What do you know about the mainboard of the Bambu? It has some custom components such as the LIDAR camera that we would not be likely to be able to support, and the accelerometer would also require entirely new code. The LCD controller is also entirely new to us. So it's going to be a long road to supporting the Bambu. -- @thinkyhead

In terms of the mainboard.. not much, makerspace management gets really nervous when i go anywhere near the printer with a screwdriver atm, i try to get more information on it as soon as i can though. (help-wanted: get high resolution images of the mainboard and AMS board?)

From my experiments yesterday i don't think that the LIDAR does anything beyond marketing BS at least as of yet as it seems that Bambulabs handles super majority of all functionality though basic GCODE (see https://github.com/Ultimaker/Cura/pull/15474).

So hypothesis is that being able to interpret GCODE for movement, extruder and heatbed will probably get better printing results already.

Accelerometer also doesn't seem to work as shaking the printer during it's configuration doesn't seem to affect the print in any way.

LCD i guess doesn't have to work or can be replaced with a compatible one if needs be.