MarlinFirmware/Marlin #25802: [BUG] MKS Robin Nano v1.2 touch screen doesn't work

berkaybtr, 2023-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have marlin 2.0.9.3 firmware on my Bluer Plus working ok, I'm trying to install newer version of marlin, I have tried all versions from 2.0.9.4 to 2.1.2 included bugfixes but I could not make my touch screen work, when I send M995 command have ghost touching and calibration fails. I was compiling 2.0.9.3 and older  versions using "robin_nano35_maple" environment without any problem. With newer versions I get errors if I compile with maple env. Sorry for my English.

### Bug Timeline

Versions after 2.0.9.3

### Actual behavior

Have ghost touching and calibration fails

### Version of Marlin Firmware

2.1.2

### Printer model

Two Trees BlueR Plus

### Electronics

MKS Robin Nano v1.2 with TFT43

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/11437853/Config.zip)

berkaybtr, 2023-05-10:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/11445717/config.zip)

berkaybtr, 2023-05-14:
Is there anyone to help me? I'm trying to make it work for 6 days, touch screen doesn't work.

ahmetcemturan, 2023-05-17:
`//
#define MKS_ROBIN_TFT35

//
// 480x272, 4.3", FSMC Display From MKS
//
//#define MKS_ROBIN_TFT43

//
// 320x240, 3.2", FSMC Display From MKS
// Usually paired with MKS Robin
//
//#define MKS_ROBIN_TFT_V1_1R

//
// 480x320, 3.5", FSMC Stock Display from Tronxy
//
//#define TFT_TRONXY_X5SA

//
// 480x320, 3.5", FSMC Stock Display from AnyCubic
//
//#define ANYCUBIC_TFT35

//
// 320x240, 2.8", FSMC Stock Display from Longer/Alfawise
//
//#define LONGER_LK_TFT28

//
// 320x240, 2.8", FSMC Stock Display from ET4
//
//#define ANET_ET4_TFT28

//
// 480x320, 3.5", FSMC Stock Display from ET5
//
//#define ANET_ET5_TFT35

//
// 1024x600, 7", RGB Stock Display with Rotary Encoder from BIQU-BX
//
//#define BIQU_BX_TFT70

//
// 480x320, 3.5", SPI Stock Display with Rotary Encoder from BIQU B1 SE Series
//
//#define BTT_TFT35_SPI_V1_0

//
// Generic TFT with detailed options
//
//#define TFT_GENERIC
#if ENABLED(TFT_GENERIC)
  // :[ 'AUTO', 'ST7735', 'ST7789', 'ST7796', 'R61505', 'ILI9328', 'ILI9341', 'ILI9488' ]
  #define TFT_DRIVER AUTO

  // Interface. Enable one of the following options:
  //#define TFT_INTERFACE_FSMC
  //#define TFT_INTERFACE_SPI

  // TFT Resolution. Enable one of the following options:
  //#define TFT_RES_320x240
  //#define TFT_RES_480x272
  //#define TFT_RES_480x320
  //#define TFT_RES_1024x600
#endif

/**
 * TFT UI - User Interface Selection. Enable one of the following options:
 *
 *   TFT_CLASSIC_UI - Emulated DOGM - 128x64 Upscaled
 *   TFT_COLOR_UI   - Marlin Default Menus, Touch Friendly, using full TFT capabilities
 *   TFT_LVGL_UI    - A Modern UI using LVGL
 *
 *   For LVGL_UI also copy the 'assets' folder from the build directory to the
 *   root of your SD card, together with the compiled firmware.
 */
#define TFT_CLASSIC_UI
//#define TFT_COLOR_UI
//#define TFT_LVGL_UI

#if ENABLED(TFT_COLOR_UI)
  /**
   * TFT Font for Color_UI. Choose one of the following:
   *
   * NOTOSANS  - Default font with antialiasing. Supports Latin Extended and non-Latin characters.
   * UNIFONT   - Lightweight font, no antialiasing. Supports Latin Extended and non-Latin characters.
   * HELVETICA - Lightweight font, no antialiasing. Supports Basic Latin (0x0020-0x007F) and Latin-1 Supplement (0x0080-0x00FF) characters only.
   */
  #define TFT_FONT  NOTOSANS

  //#define TFT_SHARED_IO   // I/O is shared between TFT display and other devices. Disable async data transfer.
#endif

#if ENABLED(TFT_LVGL_UI)
  //#define MKS_WIFI_MODULE // MKS WiFi module
#endif

/**
 * TFT Rotation. Set to one of the following values:
 *
 *   TFT_ROTATE_90,  TFT_ROTATE_90_MIRROR_X,  TFT_ROTATE_90_MIRROR_Y,
 *   TFT_ROTATE_180, TFT_ROTATE_180_MIRROR_X, TFT_ROTATE_180_MIRROR_Y,
 *   TFT_ROTATE_270, TFT_ROTATE_270_MIRROR_X, TFT_ROTATE_270_MIRROR_Y,
 *   TFT_MIRROR_X, TFT_MIRROR_Y, TFT_NO_ROTATION
 */
#define TFT_ROTATION TFT_NO_ROTATION

//=============================================================================
//============================  Other Controllers  ============================
//=============================================================================

`
This is my relevant part from my Sapphire pro.   just try it out

berkaybtr, 2023-05-17:
[ahmetcemturan](https://github.com/ahmetcemturan) I think I need to comment out "#define MKS_ROBIN_TFT35" section and  
 uncomment"TFT43" section because my display is TFT43, is it right?

berkaybtr, 2023-05-17:
I'm using marlin color ui

berkaybtr, 2023-05-17:
Now I'm able to compile firmware using robin_nano35_maple env. and touch calibration works, but as soon as marlin logo shows up on the screen, parts cooling fans start to spin and stops about 15 seconds later, but it was not with non-maple env. Whats the difference between maple and non-maple env?

ahmetcemturan, 2023-05-17:
Maple is not used anymore I think.. Dont You get a warning or so?

berkaybtr, 2023-05-17:
I think I will change the board with SKR, I'm really tired of this MKS board.

ahmetcemturan, 2023-05-18:
Just dont use maple environment.  It should give you options along with that warning.

berkaybtr, 2023-05-18:
Yes I did that already, Platformio says "use robin_nano_v1v2 instead"  but touch display doesn't work with non-maple anv.

ahmetcemturan, 2023-05-18:
Tabiiki.   If you give me your email I ca send you the whole marlin folder I am using.  Then You can check what's different..

berkaybtr, 2023-05-18:
berkayturen@hotmail.com  Thank you very much.

thinkyhead, 2023-05-30:
Any new results to report? There are some things that do only work with Maple, so we would like to get those fixed and consistent. I suspect that discrepancy has to do with timer allocation.

closed 2023-08-10 by github-actions[bot]