MarlinFirmware/Marlin #25803: [BUG] No response to G0/G1 command anymore (I suspect #25418)

rondlh, 2023-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

No response to G0/G1 command anymore.
G28/G34 executes fine, but afterwards there is a complete lack of response to any command (e.a. M119), but the temp reporting continues.

### Bug Timeline

2023-05-10 (I suspect #25418)

### Expected behavior

Response to G0/G1 commands as before

### Actual behavior

No response to G0/G1 commands which is bad for print quality

### Steps to Reproduce

Try a G0/G1

### Version of Marlin Firmware

Latest Marling 2.1.x bugfix 2023-05-10

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11438130/Configuration_adv.zip)

ellensp, 2023-05-10:
G0/G1 works as fine with stock config on ramps on current bugfix 

Provided config wont even compile on stock marlin due to 
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1852:40: error: floating constant in preprocessor expression
   #define Z_CLEARANCE_BETWEEN_PROBES   1.5 // Z Clearance between probe points

rondlh, 2023-05-10:
> G0/G1 works as fine with stock config on ramps on current bugfix
> 
> Provided config wont even compile on stock marlin due to buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1852:40: error: floating constant in preprocessor expression #define Z_CLEARANCE_BETWEEN_PROBES 1.5 // Z Clearance between probe points
> 
> I cannot test with a MKS_MONSTER8_V1

I allow Z_CLEARANCE_BETWEEN_PROBES to be a float, and have disabled a few lines in Conditionals_post.h because the preprocessor cannot handle floats.
  //#if Z_CLEARANCE_BETWEEN_PROBES > Z_CLEARANCE_FOR_HOMING 
  //  #define Z_CLEARANCE_BETWEEN_MANUAL_PROBES Z_CLEARANCE_BETWEEN_PROBES
  //#else
    #define Z_CLEARANCE_BETWEEN_MANUAL_PROBES Z_CLEARANCE_FOR_HOMING
  //#endif

I also have a custom extui, so I'm not sure if you can compile it.

BTW: Marlin 2023-05-09 works fine for me, but this update gives me serious issues. So I doubt this is a hardware issue.

rondlh, 2023-05-10:
Problem seems to be on my side...

closed 2023-05-10 by rondlh