MarlinFirmware/Marlin #25806: [BUG] try to change the step/mm of Z, I get an abnormal value

colcolcheese, 2023-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

try to change the step/mm of Z, I get an abnormal value

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]

ellensp, 2023-05-10:
no configs, so no way to replicate and no way to help you.

ellensp, 2023-05-10:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/23849

colcolcheese, 2023-05-10:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11441958/Marlin.zip)
I'm sorry. I forgot.
this bugfix-2.1.x
If 2.1.x is also fixed, please close it.
I am testing with the version I downloaded 2 hours ago.

ellensp, 2023-05-10:
It still happens on bugfix, but since its been reported before, this duplicate issue must be closed.   Please add a "Im getting this also" type message to https://github.com/MarlinFirmware/Marlin/issues/23849

colcolcheese, 2023-05-10:
Thanks for your thoughtful and clear advice