MarlinFirmware/Marlin #25807: [BUG] Strange temperatures on homescreen (DGUS_LCD_UI RELOADED)

JaxTheWolf, 2023-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After upgrading my printer to the latest Marlin (from bugfix-2.1.x) the temperatures reported by the screen for both the hotend and hotbed are incorrect, around 240˚C. Heating up works fine, OctoPrint shows the correct values, so I guess it's something to do with the way the data is being sent to the display. 

### Bug Timeline

It happened today when I built Marlin from the bugfix-2.1.x branch (10 May 2023)

### Expected behavior

The temperatures shown on the display should be correct.

### Actual behavior

The temperatures on the screen are wildly off.

### Steps to Reproduce

1. Build Marlin with the RELOADED DGUS UI (bugfix-2.1.x, last commit 7707cbf338436d7e0331fc95bc82ead06ca50f90)
2. Turn on the printer
3. The temperatures reported by the screen are incorrect

### Version of Marlin Firmware

bugfix-2.1.x (May 10 2023 18:45:02)

### Printer model

Hiprecy Leo 1S

### Electronics

FYSTEC S6, dual Z drivers. Otherwise stock.

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11444432/Configuration.zip)

ellensp, 2023-05-15:
Reloaded had an update to the the required assets 

Latest Bugfix needs https://github.com/Neo2003/DGUS-reloaded/releases installed into the lcd
Older used  https://github.com/Desuuuu/DGUS-reloaded/releases installed into the lcd

Was Updated on Mar 14 https://github.com/MarlinFirmware/Marlin/pull/25490

You need to re-flash your LCD

closed 2023-05-15 by ellensp

JaxTheWolf, 2023-05-16:
@ellensp oh, didn't notice that fork! Thank you for the heads up. I'll update and close if everything's alright.