MarlinFirmware/Marlin #25809: [BUG] (bug summary)Y-Axis will only move one direction bugfix2.1.x (SKR 3 EZ (STM32H723)

iCrux, 2023-05-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm having issues with bugfix2.1.x  where my Y-axis will only move one direction. 
First I have the newest rev. of the SKR 3 EX with STM32H723Vx.btt CPU and 5x TMC2209 drivers.
The first time I flashed the board after hooking it up to my Ender3V2Neo it booted up and everything worked as expected. I did a bed mesh/leveling, printed a test print and everything was fine. There was still a few things I needed to tweak in the firmware that couldn't be changed threw the EEPROM settings. So I recompiled/flashed a second time and at first everything was fine. I homed all axis's then started to redo the mesh and it suddenly forgot how to operate the bed/ Y-axis. It moves away from the end stop no matter what I do and that is the only direction it has moved with Marlin installed since second flash.
Things I've tried;
Reset the EEPROM ten million times in between every flash/major change...
Double,tripple checked then changed out the wire and end-stop switch for Y anyway to be 100% sure...
Flashed it back to the first firmware...
Changed the location of the endstop in configuration.h
Changed the stepper motor direction  in configuration.h and tried every variation of the two on and off...
Flashed board to RepRap to test that it wasn't the board and everything worked...
Flashed back my first compiled Marlin bugfix FW over working reprap but it continued to only move y-axis in one direction no matter if I was doing the move from the LCD screen, OctoPrint connected threw USB, Pronterface threw USB, YAT or Putty using terminal it only moves that one direction now with marlin installed no matter what I do.
I've started over with fresh Bugfix2.1.x download with clean configuration.h and configuration_adv.h and at this point don't know what to do.
Currently I have RepRap configured and running perfectly with a ESP8266 threw DUET webserver but I really want Marlin running on it so I can control it with the LCD screen that I spent too much time on to get working with this board and Marlin...
I've attached my configuration.h and configuration_adv.h that will compile and flash with everything working except the Y-axis being broken.  

### Bug Timeline

Just started after installing new main board

### Expected behavior

Y-axis to move bidirectional as designed

### Actual behavior

Y-axis only moves unidirectional and does not change direction no matter the input command or configuration change. 

### Steps to Reproduce

1) Compile Bugfix2.1.x with my configuration files
2) Flash to SKR 3 EZ with default_envs = STM32H723Vx.btt (This board comes with two different CPU's)
3) Test printer everything should work right.
4) If everything is working right. 
5) Then go back to config files and enable linear advance.
6) Test printer to find Y-axis breaks shortly after homing all axis.

### Version of Marlin Firmware

bugfix2.1.x

### Printer model

Creality Ender-3 v2 Neo

### Electronics

SKR 3 EZ-STM32H723Vx.btt

### Add-ons

independently controlled duel Z-axis stepper motors 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config's.zip](https://github.com/MarlinFirmware/Marlin/files/11448419/config.s.zip)

rondlh, 2023-05-11:
Make sure you have the absolute latest bugfix, the version of a few days ago (around 2023-05-05)  had the kind of issues you describe

iCrux, 2023-05-11:
I use TortoiseSVN and check regularly for any updates or changes before compiling again. I will try again tonight when I get home from work. But I just tried last night for the millionth time right before I open up this bug report.

KwaZuluNatal, 2023-05-16:
Have you had any luck with this issue? I just got this board and am struggling with it.

tkunchick, 2023-05-22:
Same issue. x and z move fine, Y just one direction regardless of which way you send it.

SKR3 EZ with latest bug fix.

iCrux, 2023-05-27:
I was planning to try and get the ender 3 V2 screen my printer came with working with the SKR 3 EZ board.
I've already done enough tinkering to determine that it is possible to get the V2 dwin screen to work with the SKR 3/EZ 
But since I can't even use Marlin with this board right now. I just put klipper on it and used a old android TV box as my RPI alternative and I'm using a old galaxy S7 edge with klipperscreen as my printer screen for now. Although I don't need any screen since eim using klipper it was more of a cause I can thing.

thinkyhead, 2023-05-30:
I wish we had an easier method to assist you with narrowing down the cause, but really the only way to do it is through a bisect procedure.

Would you be able to test some `bugfix-2.1.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

iCrux, 2023-05-31:
> I wish we had an easier method to assist you with narrowing down the cause, but really the only way to do it is through a bisect procedure.
>
> Would you be able to test some `bugfix-2.1.x` commits to [...]

I would be willing to do some testing just to help figure this bug out. I have completely abandoned Marlin since getting Klipper working flawlessly. I also picked up a LePotato to replace the android TV box (UGOOS UM3) I had it on. But I still have everything saved as it was when I gave up on getting Marlin working. Just let me know where you want me to start. Just so you know I have tried just enabling the bare minimal for the printer to function and compiling. So without any screen, bed mesh, LA, etc.. Just configured all the steppers and endstops and compiled and it still did the same thing with the one axis.

tkunchick, 2023-06-05:
I don't really understand it all.

I found https://github.com/dmarafino/Marlin2.1.2Hotfix-CR10-SKR3/tree/main

I was able to modify this to work with the ender 3 v2 neo and dual independent z (g34)

A couple things I noticed.

**Marlin 2.1.2 with Hot Fixes**
 
 uses 
**#define MOTHERBOARD BOARD_BTT_SKR_V3_0**  

instead of   

**#define MOTHERBOARD BOARD_BTT_SKR_V3_0_EZ**

As far as I can tell, they should be using the same pins, but it's all really over my head. I don't understand how all the parts fit together for the auto build.

There are probably other differences that I just don't recognize. 

I hope this helps.

thisiskeithb, 2023-06-05:
> I found https://github.com/dmarafino/Marlin2.1.2Hotfix-CR10-SKR3/tree/main

> As far as I can tell, they should be using the same pins, but it's all really over my head. I don't understand how all the parts fit together for the auto build.

The SKR 3 & EZ use the same common pins file. Also, both that fork & current Marlin use the same `platform` & `platform_packages` versions, but I haven't combed the code to see what else is changed:

https://github.com/MarlinFirmware/Marlin/blob/70288c6c4fae6a6334d7c48b9ef17da7ecf3c173/ini/stm32h7.ini#L68-L69

It could be an issue with bad/old cached libraries or toolchain on your machine causing issue with TMCs. You can try the following to force a fresh download:

1. Delete the hidden `.pio` folder at the root of your Marlin project.

2. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

3. Launch VSCode & reinstall PlatformIO.

4. Build a fresh copy of `bugfix-2.1.x` & flash firmware.

One of my test machines is an Ender-3 with an STM32H743VIT6-based SKR 3, TMC2209s, & a BLTouch and that has not had issues with losing position randomly.

I'll get a STM32H723VGT6-based SKR 3 swapped in and see if I can reproduce the issue.

tkunchick, 2023-06-22:
> The SKR 3 & EZ use the same common pins file. Also, both that fork & current Marlin use the same `platform` & `platform_packages` versions, but I haven't combed the code to see what else is [...]

sorry I took so long to get back, I did delete everything and reinstall like you suggested today. no change in status. y still goes one direction regardless of which way I tell it.

thisiskeithb, 2023-07-01:
@tkunchick: Can you attach your configs?

This may help narrow down some common setting that I don't have enabled since I can't reproduce this issue on an SKR 3 EZ with STM32H723VGT6.

thisiskeithb, 2023-07-01:
> I'll get a STM32H723VGT6-based SKR 3 swapped in and see if I can reproduce the issue.

Since current `bugfix-2.1.x` has some serial-related issues, I've rolled back to 4a734e4de436d9f497ef432744902c12bab9a257 and could not reproduce the issue with the Y-axis only moving one direction on the STM32H723VGT6-based SKR 3/EZ with TMC2209s.

I started testing with just the SKR 3 EZ & motor connected to Y by sending back to back Y moves in both positive & negative directions and the Y motor moved as expected.

I then hooked the SKR 3 EZ up to my Ender-3 test machine and sent repeated back to back Y moves in both positive & negative directions as well as back to back `G34`/`G29` commands without issue. I then started a print that took a couple hours and the Y-axis (or any of the others) never lost position.

<details><summary>For reference, here's PIO's Packages & Dependency Graph:</summary>
<p>

```prolog
Executing task: platformio run 

Processing STM32H723VG_btt (board: marlin_STM32H723VG; platform: ststm32@15.4.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H723VG.html
PLATFORM: ST STM32 (15.4.1) > STM32H723VG (564k RAM. 1024k Flash)
HARDWARE: STM32H723VGT6 550MHz, 564KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20200.221104 (2.2.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|-- LiquidCrystal @ 1.5.1
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20230605081836
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode
```

</p>
</details>

Voxel3DPrinting, 2023-07-02:
Hi @thisiskeithb.

I have the same issue with the normal SKR3 board.

How do I go back to the version you mention?? Without the Y issue?

Voxel3DPrinting, 2023-07-03:
I am still getting the issues with 2.1.2.1 marlin.

Configurations;
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11936479/Configurations.zip)

closed 2023-09-12 by github-actions[bot]

LouDnl, 2023-10-22:
This issue just popped up for me on a BTT SKR 3 after compiling and flashing latest release and with latest bugfix.
My printer ran fine but I couldnt run PID auto for my bed so decided to compile a new version and now my Y will only move one way 🤣