MarlinFirmware/Marlin #25813: [BUG] Unable to build a marlin firmware with STM32F103VE_longer_maple environment

rbora13, 2023-05-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,
For 2 days, I and others have not been able to build a Marlin firmware with STM32F103VE_longer_maple environement and non bugfix releases. With the latest bugfix version, compilation is fine.
The problem is that I have many problems with the non maple environment  (reboot when X endstop is closed,...)
So, is there a way to build Marlin firmware with maple environment and reliable/not bugfix marlin sources?
Thanks for your help

### Bug Timeline

issue started 2 days ago. Before that day, with exactly the same sources (V2.1.2) everything was fine

### Expected behavior

I expected to build a Marlin firmware with maple environment

### Actual behavior

but I receive this error message:
 ```Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\tr\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\tr\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\tr\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\tempMarlin\Marlin-2.1.x_20230512\buildroot\share\PlatformIO\scripts\custom_board.py", line 14:
    marlin.relocate_firmware(address)
  File "C:\tempMarlin\Marlin-2.1.x_20230512\buildroot\share\PlatformIO\scripts\marlin.py", line 26:
    replace_define("VECT_TAB_ADDR", address)
  File "C:\tempMarlin\Marlin-2.1.x_20230512\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:
==================================================================================== [FAILED] Took 11.29 seconds ====================================================================================

Environment               Status    Duration
------------------------  --------  ------------
STM32F103VE_longer_maple  FAILED    00:00:11.293
=============================================================================== 1 failed, 0 succeeded in 00:00:11.293 =============================================================================== ```

### Steps to Reproduce

1. download Marlin v2.1.2 and config files for alfawise U20
2. choose STM32F103VE_longer_maple environment in platformio.ini file
3. build sources

### Version of Marlin Firmware

all versions except Marlin bugfix

### Printer model

alfawise U20 / longer LK1

### Electronics

stock 3d printer

### Add-ons

no add-ons

### Bed Leveling

None

### Your Slicer

IdeaMaker

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.2 standard U20 config files.zip](https://github.com/MarlinFirmware/Marlin/files/11461725/Marlin-2.1.2.standard.U20.config.files.zip)

ellensp, 2023-05-12:
Python is really hating the following form buildroot/share/PlatformIO/scripts/marlin.py

```PYTHON
def replace_define(field, value):
    for define in env['CPPDEFINES']:
        if define[0] == field:
            env['CPPDEFINES'].remove(define)
    env['CPPDEFINES'].append((field, value))

```

replace it with

```PYTHON
def replace_define(field, value):
    found = ''
    for define in env['CPPDEFINES']:
        if define[0] == field:
           found = define
    if found != '':
        env['CPPDEFINES'].remove(found)
    env['CPPDEFINES'].append((field, value))
```

rbora13, 2023-05-12:
Thank you very much for your quick response. Now, I am able to build a firmware but, after flashing, unfortunately screen is blank :(

ellensp, 2023-05-12:
previously reported https://github.com/MarlinFirmware/Marlin/issues/25698
As a test if you disable  #define SDSUPPORT   does it now boot?

rbora13, 2023-05-12:
Unfortunately, same result: no boot, with or without #define SDIO_SUPPORT disabled

ellensp, 2023-05-12:
#define SDSUPPORT != #define SDIO_SUPPORT

rbora13, 2023-05-12:
Of course ;-) I meant, I tried both : 1rst test SDSUPPORT disabled, 2nd test SDSUPPORT  **AND** SDIO_SUPPORT disabled

bulat1304, 2023-05-13:
the same problem for version 2.0.x

ellensp, 2023-05-16:
Latest bugfix and platformio I can no longer replicate this issue and code above is not changed

thisiskeithb, 2023-05-17:
> Latest bugfix and platformio I can no longer replicate this issue and code above is not changed

I can't reproduce it either. We also use this config in CI tests and those are passing.

closed 2023-05-17 by thisiskeithb

rbora13, 2023-05-18:
Hi,
Of course the issue doesn't exist with the latest bugfix, that's the first sentence of my first message. The problem is that the latest bugfix version is full of bugs. So, if I understand correctly, I have to wait the next stable release?

Q07707, 2023-05-25:
I also encountered the same problem as you, how did I solve it?  My environment is  tronxy-stm32f446
Just replacing a portion of the source code in buildroot/share/PlatformIO/scripts/marlin.py?
But can the firmware run normally?

rbora13, 2023-05-27:
I’m pretty sure not. If I understand correctly, you have three options:

1. choose non maple environment
2. wait for the next stable release
3. move to klipper
Personally, I chose the 3rd one and everything is perfect! XD

ellensp, 2023-05-29:
This was fixed here 
https://github.com/MarlinFirmware/Marlin/commit/c2decc3e2e30c7cb0f517b7e40d8138a8c1d4a81

-    for define in env['CPPDEFINES']:
+    envdefs = env['CPPDEFINES'].copy()
+    for define in envdefs:

Q07707, 2023-05-29:
Thank you, I can compile successfully, the firmware runs normally.