MarlinFirmware/Marlin #25814: [FR] Add option to turn Input Shaping ON an OFF by gcode

Tupson444, 2023-05-12:
### Describe the feature you want

I'd like to be able to turn Input Shaping ON and OFF via GCODE, not only through menu. I guess it should be "M593 S1" and "M593 S0".

Also, there is no way to have Input Shaping enabled in fw but to be turned OFF by default. When I flash fw, then do Initialize EEPROM, I always have to manually turn off Input Shaping through menu (I only have it enabled for experimental purposes, and I usually don't use it).

Tupson444, 2023-05-12:
Ok, thanks. I knew that setting the frequency effectively turns it off, but since by clicking "Enable X/Y Shaping" frequencies get restored, I thought there is a different way to turn it off.

Tupson444, 2023-06-11:
I realized there is a reason to be able to simply turn Input Shaping ON and OFF via GCODE, without changing frequencies:

I do a test print and measure frequencies which are then saved. Then in some prints I want to use IS and in some not (e.g. for IS i would attach the filament spool somewhere else, while sometimes it's more convenient to just put the spool on the printer, but in this case I don't want to use IS). It would be good if I could put something to turn IS ON/OFF in the start code depending on what I print, so I don't have to manually go to menu to turn IS ON/OFF.

If I put e.g. "M593 X35 Y35" in the start code, then some time later I change something on the printer and remeasure frequencies, then I would have to change all previously generated GCODE files.

Can we reopen this issue? Or I should open a new one?

thisiskeithb, 2023-06-11:
> It would be good if I could put something to turn IS ON/OFF in the start code depending on what I print, so I don't have to manually go to menu to turn IS ON/OFF.

You’d still zero out frequencies via G-code.

Tupson444, 2023-06-11:
But when I turn IS on via menu, the frequencies get restored to 35Hz, without the need to set them. It would be good if the same could be achieved via GCODE. I know it's not possible currently, but it would be good if it was (to turn IS OFF without "erasing" set frequencies, so later can be turned ON with the frequencies restored).

thisiskeithb, 2023-06-11:
> But when I turn IS on via menu, the frequencies get restored to 35Hz (the value I had set in the fw), without the need to set them.

> It would be good if the same could be achieved via GCODE.

You can. Using your example of `35` above, you'd run `M593 XY F35` or just `M593 F35` since the frequencies match for X & Y in your example.

In regards to your original post:

> there is no way to have Input Shaping enabled in fw but to be turned OFF by default. When I flash fw, then do Initialize EEPROM, I always have to manually turn off Input Shaping through menu

I specifically asked for this ability when Tom was adding IS to Marlin and it's available.

If you want IS to be compiled in & available, but disabled at runtime, set `SHAPING_FREQ_X`/`SHAPING_FREQ_Y` in `Configuration_adv.h` to `0` or something well below your standard frequencies since a value `0` will use up a ton of ram due to the required buffers and you may get a RAM overflow error while compiling.

Then, Input Shaping would always be available, but working below any frequency that would affect your print.

If you want to run a print with IS, then you'd set the appropriate frequency with `M593` using your start g-code or whatever workflow you're using.

Alternatively, you could keep your `SHAPING_FREQ_X`/`SHAPING_FREQ_Y` at `35` and utilize `STARTUP_COMMANDS` to zero them out on boot.

Tupson444, 2023-06-11:
To have IS turned OFF by default, I now use auto0.g file with M593 F0. That is not a problem.

> You can. Using your example of 35 above, you'd run M593 XY F35 or just M593 F35 since the frequencies match for X & Y in your example.

I know I can set the frequency that way. But what if I have GCODE file with M593 XY F35, and then I modify hardware on the printer so the measured frequency changes to e.g. 42Hz. If I run the same GCODE file now, it would set them to 35 which would not be correct.

This is a minor issue, as I can simply do it manually (either change GCODE file, or turn IS on/off via menu), but it would be more convenient if there was a GCODE command for changing state, not just frequencies where 0=off.

thisiskeithb, 2023-06-11:
> But what if I have GCODE file with M593 XY F35, and then I modify hardware on the printer so the measured frequency changes to e.g. 42Hz. If I run the same GCODE file now, it would set them to 35 which would not be correct.

Marlin is going to process whatever g-code / value you send it. It has no way to know if you intend to use a frequency of 35, 42, or 100 since those are all valid values.

Tupson444, 2023-06-11:
Yes, but that is why I would like to have a command like "M593 S1" which would restore the last used frequencies before it was turned off by "M593 S0".