MarlinFirmware/Marlin #25819: Unenable to change the auto-alignment orientation

JPR75, 2023-05-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have added a second Z axe to my printer and recompiled Marlin to enable auto-alignment of Z axes.
The Z stepper motors are along the Y axe (back and front of the printer, back being where the stock Z stepper motor is located).
I have enabled the auto-alignment option in Configuration_adv.h:

```
#define Z_STEPPER_AUTO_ALIGN
```

Everything works except that the auto-alignment is done along the X axe.
As described in Configuration_adv.h file, I have changed the orientation:

 ```
  #ifndef Z_STEPPER_ALIGN_XY
    #define Z_STEPPERS_ORIENTATION 1  //jpr
  #endif
```

but this didn't change the orientation to Y as expected. Putting 0 instead of 1 does not change the orientation neither.
I have also tried different combinations with Z_STEPPER_ALIGN_XY instead of using Z_STEPPERS_ORIENTATION, but the auto-alignment keep going along the X axe.
To finish, I tried to change the orientation with M422 command through Octoprint without success.

Note: apart the orientation problem, auto-alignment seems to work correctly because once it is done on the X axe, my bed is completely de-leved with one side up and the other down.

[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/11469286/cfg.zip)

### Expected behavior

Setting the option:

 ```
  #ifndef Z_STEPPER_ALIGN_XY
    #define Z_STEPPERS_ORIENTATION 1  //jpr
  #endif
```

would change the auto-alignment from X axe to Y axe

### Actual behavior

Auto-alignment is done along the X axe regardless of Z_STEPPERS_ORIENTATION, Z_STEPPER_AUTO_ALIGN and M422 values.

### Steps to Reproduce

1. compile Marlin with the option #define Z_STEPPERS_ORIENTATION 1
2. flash the new firmware to the board, check that the firmware where updated (firmware.bin changed to FIRMWARE.CUR in the flash disk)
3. send G34
4. check auto-alignment orientation
5. compile Marlin with the option #define Z_STEPPERS_ORIENTATION 0
6. flash the new firmware to the board, check that the firmware where updated (firmware.bin changed to FIRMWARE.CUR in the flash disk)
7. send G34
8. check auto-alignment orientation
9. compile Marlin with different combinations of values for Z_STEPPER_ALIGN_XY
10. flash ; send G34 ; check orientation ...
11.  from Octoprint terminal send different combinations of  M422 S1 X Y and M422 S2 X Y
12. send G34 ; check orientation ...

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Creality Ender 5

### Electronics

Bigtree Tech SKR v1.4 Turbo

### Add-ons

BLTouch ; dual Z axes

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-14:
Looking at the code this does not seem possible 

relevant lines
```CPP
      #elif Z_STEPPERS_ORIENTATION
        { X_CENTER, probe.min_y() }, { X_CENTER, probe.max_y() }
      #else
        { probe.min_x(), Y_CENTER }, { probe.max_x(), Y_CENTER }
      #endif
```

if Z_STEPPERS_ORIENTATION is  set to 1, then         { X_CENTER, probe.min_y() }, { X_CENTER, probe.max_y() }  is used

Are you sure the uploads are working?

To check M115 gives the date and time of the last firmware build?

GMagician, 2023-05-14:
have you tryed to comment out Z_STEPPER_ALIGN_XY? it seems that Z_STEPPERS_ORIENTATION works only when it's not defined

GMagician, 2023-05-14:
Don't consider my "erased" comment. I missed your attached config

JPR75, 2023-05-14:
Thank you for reply.
M115 gives the date:
```
FIRMWARE_NAME:Marlin 2.1.2 (May 12 2023 14:51:12)
```

I just downloaded again the bugfix-2.1.x, compiled and flashed to the printer. Now M115 gives the date:
```
FIRMWARE_NAME:Marlin 2.1.2 (May 14 2023 10:52:35)
```

It does not fixed the problem.

Note: I have wired Z2 on E2 connector on my board as specified in pins_postprocess.h:

```
// The Z2 axis, if any, should be the next open extruder port
```

Could it be that E2 needs some more configuration to work as Z2 (I didn't find anything in configuration.h and configuration_adv.h or I missed something)?

JPR75, 2023-05-14:
I tried by uncommenting #define E1_DRIVER_TYPE TMC2209 in configuration.h, but didn't change anything.

thisiskeithb, 2023-05-14:
> I just downloaded again the bugfix-2.1.x, compiled and flashed to the printer. Now M115 gives the date:
>
> FIRMWARE_NAME:Marlin 2.1.2

This should say: `FIRMWARE_NAME:Marlin bugfix-2.1.x` if you’re using bugfix.

JPR75, 2023-05-14:
> This should say: FIRMWARE_NAME:Marlin bugfix-2.1.x if you’re using bugfix.

I am using the 

```
Marlin Patched Source / Latest 2.1.x with Hotfixes / Marlin 2.1.2 with bug fixes
```

file (https://github.com/MarlinFirmware/Marlin/archive/2.1.x.zip) Not the "Marlin Nightly Source" as this one may be unstable.
In the file Marlin-2.1.x\Marlin\src\inc\Version.h, I can see:

```
#define SHORT_BUILD_VERSION "2.1.2"
```

```
#define STRING_DISTRIBUTION_DATE "2022-12-17"
```

```
#define MARLIN_HEX_VERSION 02010200
```

thisiskeithb, 2023-05-14:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

JPR75, 2023-05-14:
I have found a workaround.
In file "G34_M422.cpp" ligne 131, I have changed:

```
if (parser.seen('R')) z_stepper_align.reset_to_default();
```

by

```
z_stepper_align.reset_to_default();
```
and this inverted the axe!
I don't know what the big R of "if (parser.seen('R'))" is, but I guess that an initialization is missing.

@thisiskeithb I have not yet tested the bugfix your pointed me to, but their is the same if() test in the G34_M422.cpp file. I will try to test it next.

ellensp, 2023-05-14:
M422 R   Reset alignment and known points to the defaults. This will also be done by [M502]  
https://marlinfw.org/docs/gcode/M422.html

JPR75, 2023-05-14:
So I should have done a M422 R to clean-up some old references before using G34. Shame on me!
Maybe this can be more explicitly mentioned in G34 documentation.
I can confirm that rebuilding with to the unmodified G34_M422.cpp file, the orientation is now correct. But I think that I will stay with my hacked firmware for now, so if something goes wrong the reset will be always done.