MarlinFirmware/Marlin #25821: [BUG] Software Endstops throwing errors

expwn, 2023-05-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Should be one of these commits, not sure which. Will be same one as #25820
79b0f3158556933699cab4c5ed4e8e798b4ef4d8
5f89ef99e962a0c8bed7115c87ae4e545ca143ad
01f5bd333055e919677d95e05bec1f5f067374a4

Unsure what went wrong here, it was compiling just fine for a couple hours and then it started throwing errors on a 2nd copy of the same one that worked. Firmware fails to compile and throws errors shown in additional information.

### Bug Timeline

New

### Expected behavior

Expected firmware to compile

### Actual behavior

It did not.

### Steps to Reproduce

Use provided configs, attempt to compile.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender 3 Pro

### Electronics

SKR Mini E3 V3

### Add-ons

CR Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11469535/Marlin.zip)

Marlin\src\HAL\STM32\../../inc/../core/macros.h:182:22: note: in definition of macro '_CAT'
  182 | #define _CAT(a,V...) a##V
      |                      ^
Marlin\src\HAL\STM32\../../inc/../core/macros.h:191:44: note: in expansion of macro 'CAT'
  191 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                            ^~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:192:33: note: in expansion of macro '_ENA_1'
  192 | #define _DIS_1(O)           NOT(_ENA_1(O))
      |                                 ^~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:138:43: note: in expansion of macro '_DIS_1'
  138 | #define _DO_2(W,C,A,B)     (_##W##_1(A) C _##W##_1(B))
      |                                           ^
Marlin\src\HAL\STM32\../../inc/../core/macros.h:177:28: note: in expansion of macro '_DO_2'
  177 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:178:28: note: in expansion of macro '__DO_N'
  178 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:179:29: note: in expansion of macro '_DO_N'
  179 | #define DO(W,C,V...)       (_DO_N(W,C,NUM_ARGS(V),V))
      |                             ^~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/../HAL/STM32/../shared/Marduino.h:39:24: note: in expansion of macro 'DO'
   39 | #define DISABLED(V...) DO(DIS,&&,V)
      |                        ^~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:223:29: note: in expansion of macro 'DISABLED'
  223 | #define NONE(V...)          DISABLED(V)
      |                             ^~~~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:279:9: note: in expansion of macro 'NONE'
  279 |   #elif NONE(MIN_SOFTWARE_ENDSTOP_X, MIN_SOFTWARE_ENDSTOP_Y)
      |         ^~~~
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:52,
                 from Marlin\src\HAL\STM32\usb_host.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:288:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  288 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:52,
                 from Marlin\src\HAL\STM32\fastio.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:288:6: error: #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
  288 |     #error "MAX_SOFTWARE_ENDSTOPS requires at least one of the MAX_SOFTWARE_ENDSTOP_[XYZ] options."
      |      ^~~~~
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_flash.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\Servo.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_wired.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_sram.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_if_iic.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\sdio.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\timers.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\usb_host.cpp.o] Error 1
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\fastio.cpp.o] Error 1

ellensp, 2023-05-13:
this config is just broken in so many ways

You have commented out #if ENABLE and #endif lines
This breaks marlin as the defines within the block are all set when they re not meant to be 

You have added numbers where they are not meant to be

eg
  #define MIN_SOFTWARE_ENDSTOP_X -6
  #define MIN_SOFTWARE_ENDSTOP_Y -16
  #define MIN_SOFTWARE_ENDSTOP_Z 0

there are either 
  #define MIN_SOFTWARE_ENDSTOP_X 
  #define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z

or

  //#define MIN_SOFTWARE_ENDSTOP_X
  //#define MIN_SOFTWARE_ENDSTOP_Y
  //#define MIN_SOFTWARE_ENDSTOP_Z

ellensp, 2023-05-13:
Corrected configs builds fine

closed 2023-05-13 by ellensp