MarlinFirmware/Marlin #25822: [FR] Make it possible to save macros to EEPROM

Tupson444, 2023-05-13:
### Describe the feature you want

Currently macros are not saved to EEPROM (at least that's what it says in configuration_adv.h), and I guess it should stay that way as the default.

But it would be good to have an option to enable in the firmware to save the defined macros to EEPROM with M500.

An example use:
I'm using two different bed plates for my printer. One is glass which doesn't require mesh bed leveling, and the other is flexible magnetic one, which is not flat. When I switch plates, I run a custom command (CUSTOM_MENU_MAIN) which changes PID settings and turns on/off mesh bed leveling. However, when the printer is rebooted, bed leveling is turned off. I know I can put M420 S1 in the start code in the slicer, but I don't want to because I also use glass plate which is flat.

As a workaround, when I switch plates I add "auto0.g" file to the SD card with M420 S1. But if macros were saved to EEPROM, I could put a macro in start code, which would be defined by custom command only once I switch bed plate.

PendulumPit, 2023-05-13:
Use Auto0.g to keep your macros.

https://www.youtube.com/watch?v=fpphvl5C2Cw

Tupson444, 2023-05-13:
As I said, I already do use auto0.g, but it would be more convenient if there was an option to save it to EEPROM.

ilko-k, 2023-06-10:
use the startup commands feature in _configuration_adv.h_ : 

#define STARTUP_COMMANDS "M420 S1"

Tupson444, 2023-06-10:
> use the startup commands feature in configuration_adv.h :

I already explained why that is not a solution:

>... I know I can put M420 S1 in the start code in the slicer, but I don't want to because I also use glass plate which is flat.

I could put "M420 S1" in the start code, auto0.g, or startup command, but in all those cases whatever I put (or not put), the bed leveling cannot be turned off/on without changing the command itself. Changing the start code is not an option because of the already sliced files (that is, I don't want to slice everything two times: with and without M420 S1), startup command requires reflashing firmware, so I ended up using auto0.g file as it is the easiest to change quickly (whenever I change build plate I have to change the auto0.g file on the SD card, which is not a big deal, but it is a workaround, not a solution).

I simply haven't found any way to permanently turn bed leveling on and off through LCD menu without changing something in the firmware or the SD card content. That is why I suggested saveable macros as that would be able to do this, which could potentially be useful for other things as well.

EvilGremlin, 2023-06-10:
Just use RESTORE_LEVELING_AFTER_G28 and don't put any G29/M420 in gcode.

Tupson444, 2023-06-10:
I do use RESTORE_LEVELING_AFTER_G28, but I use MBL, not UBL, is it different regarding saving state?

I've even added custom commands with M420 S1 and M420 S0 (because "Bed Leveling On/Off" disapears from menu, I don't know why), but after reboot the bed leveling is off even though I ran M420 S1 and M500. Although now I'm not 100% sure I did everything correct, so don't take my word (I don't remember exactly what I did, I only remember I couldn't set it to stay on). When I have time I can try it again.

EvilGremlin, 2023-06-10:
Yep, only UBL can retain state.

Tupson444, 2023-06-10:
Oh, I had no idea. Thanks, I might try that then.

Anyways, I'll keep this issue open as it might be useful for other things.

ilko-k, 2023-06-10:
Eh I didn't know either, good info. This will make me change a few things regarding MBL.

Cheers.