MarlinFirmware/Marlin #25823: [BUG] runtime error during build

riccalisci, 2023-05-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Platformio don't build the firmware for my FLSUN Q5(2021-x), i have a motherboard mks robin nano v1.3(stock).

I've changed the motherboard in the Configuration.h file to BOARD_MKS_ROBIN_NANO_V1_3_F4 and i've changed the enviroment of platformio in the platformio.ini file to mks_robin_nano_v1_3_f4 (also changed in drop down menù).

if i change motherboard to mks robin nano v1.2 there isn't error during build.

### Steps to Reproduce

-changed env in the platformio.ini file to mks_robin_nano_v1_3_f4
-changed the mobo version to "BOARD_MKS_ROBIN_NANO_V1_3_F4" in Configuration.h
-start building
-Runtime error stop the build:
`RuntimeError: deque mutated during iteration:
  File "C:\Users\ricca\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\ricca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ricca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ricca\Desktop\Marlin-2.1.2\buildroot\share\PlatformIO\scripts\offset_and_rename.py", line 28:
    marlin.replace_define('STM32_FLASH_SIZE', maximum_flash_size)
  File "C:\Users\ricca\Desktop\Marlin-2.1.2\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:`

### Version of Marlin Firmware

2.1.2

### Printer model

Flsun Q5

### Electronics

MKS robin nano v1.3

### Add-ons

TMC226 UART

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

riccalisci, 2023-05-13:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11470186/Configuration.zip)

thisiskeithb, 2023-05-13:
> #define CONFIGURATION_H_VERSION 02010200

You did not test this with the latest version of Marlin.

> #define CONFIG_EXAMPLES_DIR "delta/FLSUN/Q5-nano_v1"

Using the current `bugfix-2.1.x` config from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/delta/FLSUN/Q5-nano_v1, this builds fine:

<details><summary>details:</summary>
<p>

```prolog
Linking .pio/build/mks_robin_nano_v1v2/firmware.elf
Checking size .pio/build/mks_robin_nano_v1v2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  33.6% (used 22044 bytes from 65536 bytes)
Flash: [====      ]  44.0% (used 230472 bytes from 524288 bytes)
Building .pio/build/mks_robin_nano_v1v2/firmware.bin
encrypt([".pio/build/mks_robin_nano_v1v2/firmware.bin"], [".pio/build/mks_robin_nano_v1v2/firmware.elf"])
==================================== [SUCCESS] Took 119.66 seconds ====================================

Environment          Status    Duration
-------------------  --------  ------------
mks_robin_nano_v1v2  SUCCESS   00:01:59.664
===================================== 1 succeeded in 00:01:59.664 =====================================
```

</p>
</details>

---
Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.
---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-05-13 by thisiskeithb

riccalisci, 2023-05-13:
Seem that you build for mks robin nano v1.2, right? I have a problem with v1.3 "BOARD_MKS_ROBIN_NANO_V1_3_F4"

thisiskeithb, 2023-05-13:
<details><summary>mks_robin_nano_v1_3_f4 also builds fine:</summary>
<p>

```prolog
Linking .pio/build/mks_robin_nano_v1_3_f4/firmware.elf
Checking size .pio/build/mks_robin_nano_v1_3_f4/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  27.9% (used 36572 bytes from 131072 bytes)
Flash: [==        ]  21.8% (used 228804 bytes from 1048576 bytes)
Building .pio/build/mks_robin_nano_v1_3_f4/firmware.bin
rename_target([".pio/build/mks_robin_nano_v1_3_f4/firmware.bin"], [".pio/build/mks_robin_nano_v1_3_f4/firmware.elf"])
==================================== [SUCCESS] Took 130.87 seconds ====================================

Environment             Status    Duration
----------------------  --------  ------------
mks_robin_nano_v1_3_f4  SUCCESS   00:02:10.868
===================================== 1 succeeded in 00:02:10.868 =====================================
```

</p>
</details>

riccalisci, 2023-05-13:
ok, i will try. thanks a lot. 
i will let you know!

riccalisci, 2023-05-15:
Works! Thank you very much!