MarlinFirmware/Marlin #25824: Compilation Errors if MMU2 Menus are enabled

hounslg, 2023-05-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Since the commit  [Improve MMU2 unload (like original MMU2S)](https://github.com/MarlinFirmware/Marlin/commit/001d1fd7cb86306cf6e25c11a91d0b2e478007bd) if you enable the MMU2_MENUS the compile fails with an error - 
```
Marlin\src\lcd\menu\menu_mmu2.cpp: In function 'void action_mmu2_load_all()':
Marlin\src\lcd\menu\menu_mmu2.cpp:50:19: error: '_mmu2_load_filament' was not declared in this scope; did you mean 'unload_filament'?
```
Disabling this setting, or reverting the commit allows you to compile.

### Bug Timeline

New bug since last week

### Expected behavior

The compile to succeed.

### Actual behavior

Compile fails

### Steps to Reproduce

1. Using latest head of Bugfix-2.1.x
2. Enable MMU2 in config.h, ensuring serial ports don't clash
3. Compile to validate set-up
4. Uncomment #define MMU2_MENUS
5. Compile - this will error
6. Revert the commit Improve MMU2 unload (like original MMU2S
7. Compile - this will succeed

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Modified Ender 3 with MMU2

### Electronics

BigTreeTech SKR Mini E3 2.0, MMU2 unit running latest Prusa firmware

### Add-ons

MMU2

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11470376/Marlin.zip)

I took the latest Ender 3 Bigtree Tech SKR Mini E2 config files from the Bugfix-2.1.x branch and made changes just to test the MMU2 config, just in case it was something odd in my original config files.  These 'vanilla' config files had the same behaviour as my customised ones.

Compiling using Visual Studio and Auto Build Marlin on Windows 11

hounslg, 2023-05-13:
Also meant to say I took a fresh copy of the bugfix-2.1.x branch and tested with this and had the same issue.

GMagician, 2023-05-13:
rename to _mmu2_load_to_feeder

PR #25825 (2023-05-13): Fix mmu2 compile error
Fix #25824
closed 2023-05-13 by thinkyhead