MarlinFirmware/Marlin #25826: [BUG] Auto Home - Inconsistent + Silent Fail

lukasradek, 2023-05-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a stock (electronics-wise) TronXY X5SA-500 Pro on which I decided to change the bed leveling probe. It uses the probe for homing Z.
[Original TronXY blue probe.](https://www.tronxy3d.com/products/tronxy-blue-auto-leveling-sensor-for-x5sa-xy-2-pro-x5sa-pro)
[New TronXY black TR probe.](https://www.tronxy3d.com/products/tronxy-black-tr-auto-leveling-sensor-3d-printer-kit)

The original probe worked flawlessly as expected.

After switching to the new one (no firmware change), I have noticed quite inconsistent and failing behaviour.

Basically there are 4 ways to trigger the homing procedure (on `COLOR_UI`).
**Group A:** 
Send G28 via Pronterface 
Info screen -> Menu -> Motion -> Auto Home 

**Group B:**
Info screen -> Motion GUI -> Home (house icon) 
Info screen -> Menu -> Motion -> Tramming Wizard (homing is done on wizard startup) 

With the new probe, I can perform **any number (I have tried 5) of consecutive homings that are initiated via _Group A_** and it works just fine.

**The first use of _Group B_ trigger ends up in the following:**
X, Y homes fine
Z homing first lowers the bed away from probe as expected.
But then instead of starting to move the bed towards the probe to find home, the next move (there is a visible split second stop) is another bed lowering away from the probe.
And that is it. The printer then behaves as if the homing was successful even though there wasn't any move towards the probe and the probe was never triggered (there is a trigger LED on the probe).

**Any subsequent use of _Group A_ or _Group B_ triggers end up in the following:**
X, Y homes fine
Z homing first lowers the bed away from probe as expected. 
Starts to move the bed towards the probe to find home, as expected. 
Probe triggers, bed starts moving away from probe to prepare for re-bump, as expected. 
But it never starts moving towards the probe again (for the re-bump). It just stops after the pre-re-bump back off move. 
The printer again behaves as if the homing was successful.

**Simply put...** after printer boot up I can successfully home using the _Group A_ triggers "unlimited" number of times (I have tried 5). Once I use _Group B_ trigger, it stops working for any trigger (with the first fail behaviour being different from the subsequent ones). Printer reboot puts me back on the beginning again.

I also include debug data from Pronterface terminal, but expect 2 "busy processing" logs, they are the same for all cases.
I have included logs from successful _Group A_ homing, logs from the first _Group B_ fail and logs from the subsequent fail.

### Bug Timeline

I discovered it on 2.1.2.

### Expected behavior

Successful and consistent homing.

### Actual behavior

Inconsistent a failing homing described above.

### Version of Marlin Firmware

2.1.2

### Printer model

Tronxy X5SA-500 Pro

### Electronics

Chitu V6 board (stock)

### Add-ons

Tronxy TR Black sensor (see description)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[evidence.zip](https://github.com/MarlinFirmware/Marlin/files/11472056/evidence.zip)

thinkyhead, 2023-05-30:
You reported that the problem exists in 2.1.2 and also that you tested the latest bugfix-2.1.x, but there are some big changes in bugfix-2.1.x and you didn't mention if there were any changes in behavior. It would be helpful to know what if any differences you observed.

Meanwhile, I'll have a look at the log you posted and see about recruiting the contributor who assisted with the latest probing changes.

jerrygrey, 2023-07-02:
I also ran into this bug yesterday with bugfix-2.1.x on Ender 3 and SKR-MINI-V3. It disappeared when downgrading to 2.0.x

lukasradek, 2023-07-04:
We have something!!

These are 2 serial logs from the same machine with the SAME HARDWARE (including the Z probe).

This homing was initiated from G28 command using Pronterface.

```
>>> g28
SENDING:G28
echo:G28
>>> G28  X-9.00 Y-6.00 Z0.00
Machine Type: CoreCartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X-40.00 Y0.00 Z0.00 (Left-Center & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
>>> set_bed_leveling_enabled  X-9.00 Y-6.00 Z0.00
<<< set_bed_leveling_enabled  X-9.00 Y-6.00 Z0.00
remember_feedrate_scaling_off: fr=66.67 100%
Raise Z before homing:
do_z_clearance(5.00 [0.00 to 5.00], 0)
do_blocking_move_to_z(5.00, 8.00)
>>> do_blocking_move_to  X-9.00 Y-6.00 Z0.00
>  X-9.00 Y-6.00 Z5.00
<<< do_blocking_move_to  X-9.00 Y-6.00 Z5.00
>>> homeaxis(X)
Homing bump is planned0Length mm-5.00
Home Fast: -778.50mm
>>> do_homing_move  X-9.00 Y-6.00 Z5.00
...(X, -778.50, [50.00])
is_home_dir 1
echo:busy: processing
echo:busy: processing
echo:busy: processing
ENDSTOP TRIGGERED planner fn
<<< do_homing_move  X-9.00 Y-6.00 Z5.00
Move Away: 5.00mm
>>> do_homing_move  X-9.00 Y-6.00 Z5.00
...(X, 5.00, [50.00])
is_home_dir 0
<<< do_homing_move  X-9.00 Y-6.00 Z5.00
Re-bump: -10.00mm
>>> do_homing_move  X-9.00 Y-6.00 Z5.00
...(X, -10.00, 25.00)
is_home_dir 1
ENDSTOP TRIGGERED planner fn
<<< do_homing_move  X-9.00 Y-6.00 Z5.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X-9.00 Y-6.00 Z5.00 :
<<< set_axis_is_at_home(X)
  current_position= X-9.00 Y-6.00 Z5.00 : sync_plan_position
  current_position= X-9.00 Y-6.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Homing bump is planned0Length mm-5.00
Home Fast: -759.00mm
>>> do_homing_move  X-9.00 Y-6.00 Z5.00
...(Y, -759.00, [50.00])
is_home_dir 1
echo:busy: processing
echo:busy: processing
echo:busy: processing
ENDSTOP TRIGGERED planner fn
<<< do_homing_move  X-9.00 Y-6.00 Z5.00
Move Away: 5.00mm
>>> do_homing_move  X-9.00 Y-6.00 Z5.00
...(Y, 5.00, [50.00])
is_home_dir 0
<<< do_homing_move  X-9.00 Y-6.00 Z5.00
[... 224 lines cut ...]
  current_position= X290.00 Y250.00 Z0.00 :
<<< set_axis_is_at_home(Z)
  current_position= X290.00 Y250.00 Z0.00 : sync_plan_position
  current_position= X290.00 Y250.00 Z0.00 : > AFTER set_axis_is_at_home
  current_position= X290.00 Y250.00 Z0.00 : Probe::set_deployed
deploy=0 no_return=0
>>> Probe::probe_specific_action  X290.00 Y250.00 Z0.00
<<< Probe::probe_specific_action  X290.00 Y250.00 Z0.00
>>> do_blocking_move_to  X290.00 Y250.00 Z0.00
>  X290.00 Y250.00 Z0.00
<<< do_blocking_move_to  X290.00 Y250.00 Z0.00
<<< homeaxis(Z)
<<< home_z_safely  X290.00 Y250.00 Z0.00
  current_position= X290.00 Y250.00 Z0.00 : sync_plan_position
>>> do_move_after_z_homing  X290.00 Y250.00 Z0.00
<<< do_move_after_z_homing  X290.00 Y250.00 Z0.00
restore_feedrate_and_scaling: fr=66.67 100%
<<< G28  X290.00 Y250.00 Z0.00
```

The difference are the extra "ENDSTOP TRIGGERED planner fn" logs that I have added into the `void Planner::endstop_triggered(const AxisEnum axis)` function.

Meaning the issue is caused by the fact that the "re-bump" logic recieves endstop triggered immediately (as if they were buffered??). The probe itself is not triggered though, judging by its LED and checking the signal wire with multimeter.

@thinkyhead

lukasradek, 2023-09-03:
Still a problem.

closed 2023-11-17 by github-actions[bot]