MarlinFirmware/Marlin #25830: [BUG] Wrong Extruder numbering starting from E1 in temperature.cpp

rondlh, 2023-05-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In temperature.cpp, the procedure Temperature::set_heating_message
If there is only 1 hotend, then it calls this hotend "E1", shouldn't that be "E0"?
If there are multiple hotends, then they are numbered starting "E1" for the first hotend, shouldn't counting start from "E0"?
In various other places the first extruder is called "E0", just check in Configuraton.h

```
    void Temperature::set_heating_message(const uint8_t e, const bool isM104/*=false*/) {
      const bool heating = isHeatingHotend(e);
      ui.status_printf(0,
        #if HAS_MULTI_HOTEND
          F("E%c " S_FMT), '1' + e
        #else
          F("E1 " S_FMT)
        #endif
        , heating ? GET_TEXT(MSG_HEATING) : GET_TEXT(MSG_COOLING)
      );

      if (isM104) {
        static uint8_t wait_e; wait_e = e;
        ui.set_status_reset_fn([]{
          const celsius_t c = degTargetHotend(wait_e);
          return c < 30 || degHotendNear(wait_e, c);
        });
      }
    }

```

### Bug Timeline

Note sure

### Expected behavior

Consistent use of the hotend/extruder name.

### Actual behavior

Mostly the "E0" is used for the first hotend/extruder, but in Temperature.cpp the first hotend/extruder is called "E1"

### Steps to Reproduce

Check temperature.cpp and Configuration.h

### Version of Marlin Firmware

Latest bugfix

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11475347/Configuration_adv.zip)

ellensp, 2023-05-15:
computers like starting form 0, ie marlin code, humans like starting from 1, eg in user interfaces.

If you really want this there set #define  NUMBER_TOOLS_FROM_0  Will only work on some User interfaces

closed 2023-05-15 by ellensp

rondlh, 2023-05-15:
@ellensp 
> computers like starting form 0, ie marlin code, humans like starting from 1, eg in user interfaces.
> 
> If you really want this there set `#define NUMBER_TOOLS_FROM_0` Will only work on some User interfaces

OK, thanks, I understand.
To me this seems inconsistent, at least if you use an IDEX system, you use "T0" to select E1 and the E1 temperature is reported as T0. Setting the hotend temperature with M104 also uses T0 for "E1" and T1 for "E2", isn't that inconsistent?
Enabling/disabling the "E1" stepper requires command M17/M18 E0.

Additionally, #define NUMBER_TOOLS_FROM_0 has no impact on the piece of code I mentioned.