MarlinFirmware/Marlin #25833: [BUG] Marlin Simulator on Windows compilation fail due to multiple defs

lukasradek, 2023-05-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to compile Marlin with default simulator config on Windows 11 and I am getting this linker error.

```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: 

.pio/build/simulator_windows/debug/lib9ed/libMarlinSimUI.a(application.o): in function `get_pin_mode(short)':

<path-to-marlin>\Marlin/Marlin/src/pins/../HAL/NATIVE_SIM/pinsDebug.h:41: multiple definition of `get_pin_mode(short)'; 

.pio/build/simulator_windows/debug/src/src/gcode/config/M43.o:
<path-to-marlin>\Marlin/Marlin/src/gcode/config/../../pins/../HAL/NATIVE_SIM/pinsDebug.h:41: first defined here
```

It comes from the precompiled MarlinSimulator files, so that might be in @p3p realm.

### Bug Timeline

I discovered it just now, but it must have started very recently. I guess it is a consequence of the 3rd party dependencies rework.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Simulator Windows

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-15:
disable #define PINS_DEBUGGING

lukasradek, 2023-05-15:
> disable #define PINS_DEBUGGING

Well... read through the post 😉.

p3p, 2023-05-15:
This has always been the case as I hook into Marlins pin debug feature to get the pin naming data for the sims UI, not sure of a workaround other than disabling the SimUI feature to use the Marlin feature. 

Marlin was never designed for the way the simulator uses it, but by design the sim needs to work without modifying Marlin.

lukasradek, 2023-05-15:
I am not sure either 🙂.

But isn't this just the "not matching source versions being compiled together" problem? I have never had this happen before and I actually discovered this on my long-running config that I haven't changed for quite a while (apart from version bumps).

I am not sure what you mean by _disabling the SimUI feature to use the Marlin feature_, but that probably doesn't matter since that is not in my competence to change 🙂. Wouldn't it be a viable solution though to let Marlin "publish" some "interface" file that would define that naming data but wouldn't clash with other includes inside Marlin?

ellensp, 2023-05-16:
You miss understand.

The simulator uses its own function called get_pin_mode to tap into marlin to get the pin naming data.

You cannot enable PINS_DEBUGGING in Marlin. It is simply incompatible with the way the emulator is implemented.

thinkyhead, 2023-05-30:
I patched this duplicate function problem recently. Please give it another go with the latest code and see if I got them all.

closed 2023-07-16 by lukasradek