MarlinFirmware/Marlin #25836: Compilação

Everaldo91, 2023-05-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Processing mks_robin_nano_v1_3_f4 (board: marlin_STM32F407VGT6_CCM; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407VGT6_CCM.html
PLATFORM: ST STM32 (12.1.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\Evera\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\Evera\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Evera\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Evera\Downloads\Marlin-2.0.9.6\Marlin-2.0.9.6\buildroot\share\PlatformIO\scripts\offset_and_rename.py", line 28:
    marlin.replace_define('STM32_FLASH_SIZE', maximum_flash_size)
  File "C:\Users\Evera\Downloads\Marlin-2.0.9.6\Marlin-2.0.9.6\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:
================================================= [FAILED] Took 15.85 seconds =================================================

Environment             Status    Duration
----------------------  --------  ------------
mks_robin_nano_v1_3_f4  FAILED    00:00:15.854
============================================ 1 failed, 0 succeeded in 00:00:15.854 ============================================ 

### Version of Marlin Firmware

2.0.9.6

### Printer model

two trees bluer

### Electronics

mks robin nano v1.3s

### Add-ons

stm32f407

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-16:
symptoms are the same as https://github.com/MarlinFirmware/Marlin/issues/25813#issuecomment-1545652043

Something python related...  

But although I could replicate the issue I linked. I cannot replicate it with mks_robin_nano_v1_3_f4

thisiskeithb, 2023-05-17:
No issues compiling with `BOARD_MKS_ROBIN_NANO_V1_3_F4` + `mks_robin_nano_v1_3_f4` env here. Maybe something elsewhere in the toolchain was updated?

closed 2023-05-17 by thisiskeithb

thinkyhead, 2023-07-17:
We did fix the "deque" problem in our scripts and propagated that fix to the 1.1.x and 2.0.x branches. Obviously this fix does not affect older tagged releases.