MarlinFirmware/Marlin #25837: [BUG] filament fan incorrect control

riccalisci, 2023-05-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When the fan set to 0% led D1 (in mobo) flashes a little bit and the fan try to move. when the fan set to 100% led flashes yet and the fan not run constantly.
[video](https://youtube.com/shorts/-fy1sDWeEj8)

i commented a PB1 declaration in pins->stm32f4->pins_MKS_ROBIN_NANO_V1_3_F4.h. I don't know what it's for, but probably is in conflict with PB1 declared in pins_MKS_ROBIN_NANO_common.h.

I commented the line #define LED_PIN PB1 (as in image)  and fan work correctly

[image]
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11486955/configuration.zip)

### Expected behavior

verify solution

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Flsun Q5

### Electronics

MKS robin nano v1.3

### Add-ons

TMC226 UART

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-16:
yes ```#define LED_PIN PB1``` does look to be in error.

ellensp, 2023-05-16:
Have created a PR to fix this.

closed 2023-05-16 by ellensp