MarlinFirmware/Marlin #25839: Bed leveling presets for beginners

akinilicali, 2023-05-16:
### Is your feature request related to a problem? Please describe.

I've checked the documentation on bed leveling. It very concisely explains the basics. With novices in mind, however, can we establish a set of global presets based on bed size or if need be, on the printer model? This would go with a warning that says it applies to stock printers. As we know, more and more printers are getting 32-bit motherboard upgrades these days.
If this is impractical, we could at least have a table concisely showing the amount of memory used, and the min/max GRID_MAX_POINTS needed for say a 500x500 bed.

### Are you looking for hardware support?

No

### Describe the feature you want

Bed leveling presets such as Large = 16x16
                                              Medium = 8x8
                                              Small = 5x5
Also, a message showing the memory used.
8-bit boards may dangerously get close to their limits due to a wrong bed leveling selection.

Vertabreak, 2023-05-17:
personally i would suggest something like 1 grid point every 25mm / inch more then that feels like overkill and less could also work fine something like every 50mm / 2 inch just my 2 cents.

akinilicali, 2023-05-17:
Thank you Vertabreak. For a 500mm by 500mm bed, that would entail 20x20 points. We need to think about the memory needed to keep this information and the practicality of having to wait for 400 readings from BL Touch.  I was a bit confused about the right number of mesh spots. Because of the below warning in the Configuration.h
 #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
Now, we can apparently set it to 16x16. How limited is the implementation? And, how bad does it get when we are closer to the maximum setting? In order to shed some light, I thought it would be great to have a general guideline in the form of a table.

Vertabreak, 2023-05-17:
as i described above a 500x500 bed would be fine with a 10x10 and probably not have much to gain beyond 20x20 in this instance. i would note in the past i have successfully used a 3x3 on a 255x255 bed without any issue.

thinkyhead, 2023-05-30:
The Example Configurations do a pretty reasonable job of providing good bed leveling presets, probing margin, etc. The amount of memory used is always the same, as it is configured and allocated at compile time. A float consumes 4 bytes, so a 10x10 max grid size will use 400 bytes of SRAM at all times.

akinilicali, 2023-05-30:
Thank you for the information, Scott. I appreciate your hard work. Marlin is certainly on the right track. Keep up the good work.