MarlinFirmware/Marlin #25842: Bilinear Mesh Bed Levelling Finishes at first point BL Touch [BUG] (bug summary)

fiveorders, 2023-05-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have tried nightly builds as well but all the same. 

QGL Core XY and there is nothing wrong with the genuine BL TOuch sensor or end stops. G34 Z_Alignment works fine. But when I try to mesh bed levelling it homes and says bed levelling completed. 

### Bug Timeline

It started after 2.0.9

### Expected behavior

It should do the 3x3 bilinear mesh bed levelling.

### Actual behavior

Homes and says bed levelling completed without mesh bed levelling

### Steps to Reproduce

G29

### Version of Marlin Firmware

2.1.2

### Printer model

QGL Core XY

### Electronics

Octopus Pro v1.0 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-17:
#define NOZZLE_TO_PROBE_OFFSET { 0, 49, 0, 2}  
You probing inside a hypercube or some other > 3 dimensional reality.

ellensp, 2023-05-17:
Provide config does not com[ile.

fiveorders, 2023-05-17:
yes because it is a 4 axis machine. probably you get error on sanity check right? just disable that on sanity.h it will compile. but this error has nothing to do with that because it is working on 2.0.9

fiveorders, 2023-05-19:
The problem is about the 4th axis, and the disabled end stops. I couldn't figure out how to fix yet but closing this

closed 2023-05-19 by fiveorders