MarlinFirmware/Marlin #25850: [BUG] (Dummy Thermistor shows 0)

destruct123, 2023-05-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin works fine. 
But i can't set my dual extruder printer to my set value of extruder temperature. I'm currently running a clay printer on a SKR 3 EZ 5x TMC 5160. I need to set both extruders to 200ºC and the dummy thermistor just shows 0ºC.

### Expected behavior

Extruder's temperature as set on thermistor 999 value 200ºC

### Actual behavior

Extruder temperatures shows 0ºC

### Steps to Reproduce

Boot the printer. 

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom

### Electronics

SKR V3.0 EZ + 5 TMC5160 PRO

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2023-05-30:
Weird. Dummy Thermistor 999 should always show the temperature configured for it, as it uses a custom table. But perhaps the logic for reading the table has gone wonky. Is there an earlier version of Marlin where it works okay? We csn look at the differences in the thermistor handling since that time to figure out what change most likely broke it.

ellensp, 2023-06-09:
Did a git bisect it says it broke on 03a43786703d84f1b8c6374755b2764aeb9e04eb

ellensp, 2023-06-09:
'#define TEMP_SENSOR_0_IS_THERMISTOR 1' is missing, when it should be set as 998/999 is a thermistor table

ellensp, 2023-06-09:
A PR has been created to fix this

closed 2023-06-09 by ellensp