MarlinFirmware/Marlin #25853: [BUG] G28 raise feature broken in 2.1.2.1

lostcontrol, 2023-05-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I just upgraded my CR-10 Mini from 2.1.2 to 2.1.2.1. The first print I did with the new version ended up badly as the print head crashed into the object right after the print finished.

My Octoprint is configured to send a "G28 X0" after the print ends to home X. Up to now, the head moved up then moved to X. But with 2.1.2.1, the head now moved DOWN (instead of UP), crashing into the part. I can reproduce the behavior by sending a "G28" command to the printer. It always moves down first instead of up.

### Bug Timeline

This is new. It worked with 2.1.2 and happens now with 2.1.2.1

### Expected behavior

The head should raise when sending G28.

### Actual behavior

The head moves down first when sending G28.

### Steps to Reproduce

1. Send G28
2. See the print head goes down instead of raising before homing

### Version of Marlin Firmware

2.1.2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11511172/configuration.zip)

EvilGremlin, 2023-05-18:
check `G28 X`, that extra 0 might mess it up

RV-from-be, 2023-05-18:
Send G28X or G28X0 -> No issue with the actual bugfix !

lostcontrol, 2023-05-18:
I'll try again with `G28` tomorrow. I'm pretty sure it did it. I'll also try with `G28 Rx` to see what happens. I simply rebased my local configuration changes from 2.1.2 on top of 2.1.2.1. No conflict but I'll double-check my configuration tomorrow again.

lostcontrol, 2023-05-18:
Ok, I can reproduce it but not directly after powering up. I did several `G28`, `G28 X` and it eventually occurred. I'll try to find the minimal pattern required to trigger the bug.

lostcontrol, 2023-05-18:
You can observe the behavior on the following screencast:

[Screencast from 2023-05-19 00-11-42.webm](https://github.com/MarlinFirmware/Marlin/assets/983305/42893247-c063-4412-a1a3-8f668f1dc546)

1. I power up the printer
2. I send `G28` (this is where the screencast starts)
3. After the homing on XYZ is done, I move the print head up a bit
4. I send `G28 X`
5. As you can see on the screencast, the print head goes down before homing X

It seems to me that the print head goes down to the Z position it was after the first `G28` (or even lower).

RV-from-be, 2023-05-18:
@lostcontrol 
I can assure you that I redid your procedure on my Ender3V2 (via octoprint and via the UI Jyersui....in Jyersui code HOME X is a G28X). But no issue, my Z axis doesn't go down...I use the last Marlin Bugfix. I did this test with and without FT_Motion.

lostcontrol, 2023-05-20:
> @lostcontrol I can assure you that I redid your procedure on my Ender3V2 (via octoprint and via the UI Jyersui....in Jyersui code HOME X is a G28X). But no issue, my Z axis doesn't go down...I use the last Marlin Bugfix. I did this test with and without FT_Motion.

I can assure you that I can trigger this bug anytime with 2.1.2.1 and that it was never there before. By coincidence, there are changes in `G28.cpp` in this release...

I took a few minutes to debug this further. I added a few more prints and enabled debugging. Here is my patch for the additional log messages:

```diff
diff --git a/Marlin/src/gcode/calibrate/G28.cpp b/Marlin/src/gcode/calibrate/G28.cpp
index 839e29ecc4..c91ce6ae14 100644
--- a/Marlin/src/gcode/calibrate/G28.cpp
+++ b/Marlin/src/gcode/calibrate/G28.cpp
@@ -421,6 +421,8 @@ void GcodeSuite::G28() {
     // Check for any lateral motion that might require clearance
     const bool may_skate = seenR || NUM_AXIS_GANG(doX, || doY, || TERN0(Z_SAFE_HOMING, doZ), || doI, || doJ, || doK, || doU, || doV, || doW);
 
+    DEBUG_ECHOLNPGM("z_homing_height: ", z_homing_height);
+
     if (seenR && z_homing_height == 0) {
       if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("R0 = No Z raise");
     }
@@ -428,9 +430,11 @@ void GcodeSuite::G28() {
       bool with_probe = ENABLED(HOMING_Z_WITH_PROBE);
       // Raise above the current Z (which should be synced in the planner)
       // The "height" for Z is a coordinate. But if Z is not trusted/homed make it relative.
+      DEBUG_ECHOLNPGM("seenR: ", seenR, " !TERM(...): ", !TERN(HOME_AFTER_DEACTIVATE, axis_is_trusted, axis_was_homed)(Z_AXIS));
       if (seenR || !TERN(HOME_AFTER_DEACTIVATE, axis_is_trusted, axis_was_homed)(Z_AXIS)) {
         z_homing_height += current_position.z;
         with_probe = false;
+        DEBUG_ECHOLNPGM("z_homing_height: ", z_homing_height, " current_position.z: ", current_position.z);
       }
 
       if (may_skate) {
```

And here are the logs when I send a `G28 X` which goes down instead of up:

```
Send: G28 X
Recv: echo:G28 X
Recv: >>> G28  X189.00 Y118.00 Z8.67
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X-39.00 Y-8.00 Z-0.67 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: >>> set_bed_leveling_enabled  X189.00 Y118.00 Z8.67
Recv: <<< set_bed_leveling_enabled  X189.00 Y118.00 Z8.67
Recv: z_homing_height: 4.00
Recv: seenR: 0 !TERM(...): 0
Recv: Raise Z before homing:
Recv: >>> do_blocking_move_to  X189.00 Y118.00 Z8.67
Recv: >  X189.00 Y118.00 Z4.00
[...]
Recv: <<< do_blocking_move_to  X189.00 Y118.00 Z4.00
Recv: BLTouch from 90 to 160
Recv: BLTouch from 160 to 90
Recv: >>> homeaxis(X)
Recv: Home Fast: -450.00mm
Recv: >>> do_homing_move  X189.00 Y118.00 Z4.00
Recv: ...(X, -450.00, [20.00])
[...]
Recv: <<< do_homing_move  X189.00 Y118.00 Z4.00
Recv: Move Away: 5.00mm
Recv: >>> do_homing_move  X189.00 Y118.00 Z4.00
Recv: ...(X, 5.00, [20.00])
Recv: <<< do_homing_move  X189.00 Y118.00 Z4.00
Recv: Re-bump: -10.00mm
Recv: >>> do_homing_move  X189.00 Y118.00 Z4.00
Recv: ...(X, -10.00, 10.00)
[...]
Recv: <<< do_homing_move  X189.00 Y118.00 Z4.00
Recv: >>> set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y118.00 Z4.00 :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y118.00 Z4.00 : sync_plan_position
Recv:   current_position= X0.00 Y118.00 Z4.00 : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv:   current_position= X0.00 Y118.00 Z4.00 : sync_plan_position
Recv: X:0.00 Y:118.00 Z:4.00 E:0.00 Count X:0 Y:9440 Z:1600
Recv: <<< G28  X0.00 Y118.00 Z4.00
Recv: ok
```

As you can see from my additional logs, the line `z_homing_height += current_position.z;` is never reached because both `seenR` and `!TERN(...)` return 0. So the printer goes down to `z_homing_height` which is set to 4.0 in my case.

Just to confirm, I reverted 6ebbfa832cf and everything is back to normal. @thinkyhead could you have a look, please? Thank you.

lostcontrol, 2023-05-20:
Just to make it clear. After reverting 6ebbfa832cf, if I send `G28 X` after `G28`, the print head does not move on Z because it is already at or above `z_homing_height` I guess. If I lower the print head after `G28`, then the printer will go up to `z_homing_height` before homing X.

thinkyhead, 2023-05-20:
Thanks for the thorough logging. That helps. There was some issue with the R parameter and others with homing/probing generally that we addressed recently, some of which might have been included in the 2.1.2.1 update. I'll reexamine the G28 changes and make sure the behavior is sensible, try to get a hotfix pushed as soon as I have a handle on the issue.

DerAndere1, 2023-05-20:
I think I found the offending changes and the possible fix, see related issue https://github.com/MarlinFirmware/Marlin/issues/25854

thinkyhead, 2023-05-30:
Duplicate of #25854

closed 2023-05-30 by thinkyhead