MarlinFirmware/Marlin #25854: [BUG] G28 XY Moves Z to 0 in 2.1.2.1

theonteg, 2023-05-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After updating to 2.1.2.1 from 2.1.1 homing XY axes only, with G28 XY moves Z to 0 also...

### Bug Timeline

New bug on 2.1.2.1

### Expected behavior

Homing only X and Y axis

### Actual behavior

First it moved to Z 0 and then it homed X and Y axis

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11511312/Configuration.zip)

### Version of Marlin Firmware

2.1.2.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

RV-from-be, 2023-05-18:
No issue with the actual bugfix !

thinkyhead, 2023-05-20:
Hmm, I was pretty careful not to merge changes that affected motion, but something must have slipped through my laser fence. I'll do a full comparison of 2.1.2.1 to 2.1.2 and see if any changes stand out. Such egregious behavioral problems usually come from an obvious error of some kind. I haven't actually seen the issue myself, and may not be able to reproduce it, so if I can't find an obvious answer maybe I can recruit your help to track it down by posting potential patches to my fork.

thinkyhead, 2023-05-20:
Meanwhile I forcibly pushed a fix to the 2.1.x branch to deal with potential EEPROM changes that could cause settings corruption. That's probably not related to this movement issue, but it can never hurt to do `M502`, `M500` to make sure the settings are all good.

thinkyhead, 2023-05-20:
@theonteg — Have you also tested version 2.1.2 to see if the same issue exists? I don't see any changes between 2.1.2 and 2.1.2.1 motion code that should make any difference between the two, but maybe the issue lies with some other component.

thinkyhead, 2023-05-20:
I've posted a branch for testing to my fork, which so far just makes some adjustments to the `types.h` file based on `bugfix-2.1.x`, since I vaguely recall there may have been some obscure glitches in some of our data-types. No guarantee this makes any difference, but I did a comparison as far back as 2.1.1 and couldn't see any changes to the motion code that would cause Z to move in the wrong direction or to somehow lose its DIR state.

https://github.com/thinkyhead/Marlin/tree/test_marlin_2121_better_types

The `bugfix-2.1.x` branch is handling DIR states differently, and in my recent testing I saw some DIR issues come and go. Specifically, as I was testing on a CR-10S machine with a TH3D EZBoard and TMC2208 stepper drivers the directions were all fine until I sent `M502` to apply default settings. This has the effect of resetting TMC stepper drivers, and after `M502` the X and Y axes were moving in the wrong direction. But a couple days later as I was doing more testing the issue was no longer showing up. Nevertheless, I plan to do a lot more motion testing over the coming days so I'll keep you posted if I discover anything and if it relates to what you're seeing in 2.1.2.1.

theonteg, 2023-05-20:
I have done M502 and M500 but still the same issue.
I have just found out that I am having the same issue when selecting Auto Home from lcd menu. It does not increase Z height, home X and Y and then home Z as it should. It instead lowers Z to 0 (wherever it think it is) and then tries to home X and Y.

But when I selected Level Bed from the lcd menu, it homed all axes the proper way before starting bed leveling.

I will try my previous working version (2.1.1) and the 2.1.2 you suggested me to see if I can provide you any useful feedback.

DerAndere1, 2023-05-20:
I find this in G28.cpp:
with `HOMING_Z_WITH_PROBE` enabled, `with_probe` is `true`. 

https://github.com/MarlinFirmware/Marlin/blob/72124ed1396d3cfd0994644763c15c8c1c860f9a/Marlin/src/gcode/calibrate/G28.cpp#L428

note that the signature of that function is `void do_z_clearance(const_float_t zclear, const bool lower_allowed/*=false*/)`

so `do_z_clearance` is called with argument `lower_allowed` set to `true`:

https://github.com/MarlinFirmware/Marlin/blob/72124ed1396d3cfd0994644763c15c8c1c860f9a/Marlin/src/gcode/calibrate/G28.cpp#L439
In that case, Z will move to `Z_HOMING_HIGHT` (which is 0 by default).

This does not happen when axes are untrusted / unhomed:

https://github.com/MarlinFirmware/Marlin/blob/72124ed1396d3cfd0994644763c15c8c1c860f9a/Marlin/src/gcode/calibrate/G28.cpp#L431-L434

DerAndere1, 2023-05-20:
@thinkyhead I think the offending commit is https://github.com/MarlinFirmware/Marlin/commit/6ebbfa832cfb9542d4998595394de31b5ff6a555 . Fix is in commit

klack, 2023-05-28:
It appears after cherry picking https://github.com/MarlinFirmware/Marlin/commit/f53217b2c5bf7d61d21ee6784ec251bc7a670eb3 that "Raise Z before homing" is applied, but then is not unapplied after homing.   Causing subsequent `G28 X` commands to keep raising and raising Z.

thinkyhead, 2023-05-30:
I went ahead and applied a "hotfix" to 2.1.2.1 so please re-download the updated 2.1.2.1 and see if it solves the problem. Thanks for testing the branch I posted and confirming quickly that it solved the issue, those of you who did.

lostcontrol, 2023-06-01:
@thinkyhead just redid the same test I did in #25853 with the updated 2.1.2.1 (why not make a 2.1.2.2, updating tags is so confusing) and the behavior is now correct :+1: Thank you for your work.

theonteg, 2023-06-01:
> I went ahead and applied a "hotfix" to 2.1.2.1 so please re-download the updated 2.1.2.1 and see if it solves the problem. Thanks for testing the branch I posted and confirming quickly that it solved the issue, those of you who did.

I have just updated to the revised 2.1.2.1 and it seems to work fine!
Thank you

closed 2023-06-18 by theonteg