MarlinFirmware/Marlin #25855: G28 Homing X or Y only [BUG] (bug summary)

StephanS9, 2023-05-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When entering G28 XY the Z axis is moving down to the bed.
Axis should go up or stay in place.
Homing of the Z Axis is working as it should.
Tried the Bugfixversion but it gave me a compiling error.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11511451/Configuration.zip)

### Bug Timeline

Installed 2.1.2

### Expected behavior

Movement of X and Y Axis only

### Actual behavior

Z Moves down

https://github.com/MarlinFirmware/Marlin/assets/134008482/4f0cd37d-492f-4553-94ee-719b83d64d80

### Steps to Reproduce

G28 XY

### Version of Marlin Firmware

2.1.2.1

### Printer model

Mendel Max

### Electronics

Fysetc S6 2.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

RV-from-be, 2023-05-18:
No issue with the actual bugfix !

ellensp, 2023-05-19:
Duplicate 25854

closed 2023-05-19 by ellensp

thinkyhead, 2023-05-20:
Please add comments over on #25854. I just wanted to ask, are you seeing this with 2.1.2 also or just 2.1.2.1?