MarlinFirmware/Marlin #25857: [BUG] (bug summary)

fiveorders, 2023-05-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Mesh Bed Levelling does not function. I have two same Core XY printers, Which are almost identical; only the extruders differ. One of the extruders has a PINDA probe at the (26, 0, 0) position, and the other has BLTouch at the (0, 49, 0) position. (I replaced the BlTouch with PINDA at the same position, but it didn't make any difference)

The BED is ultra-flat, but when I start a full bed 300x300 square print right side of the bed seems high, and the nozzle starts to scratch. I tried with the PINDA probe in the same position, but the same thing happened. It drives me crazy because it does the same on MArlin 2.0.9.3 and nightly builds. I am questioning if this happens after a certain distance from the nozzle. I mean, if the 49mm distance at the back of the nozzle is too far for Mesh bed levelling? But the bet is ultra-flat, and I am %100 sure. The other printer prints are perfectly fine.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/11520379/Archive.zip)
 
This is my Start Gcode. I use this 

G90 ; Set all axes to absolute
M83 ; extruder relative mode
G28 W; home all without mesh bed level
M75;

G1 X50 Y50 Z50 F4600;
M900 K75; Set K-Factor
M84 S300;

M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

G29 ;
G1 X0 Y-5 F5600 ;
G1 Z0.3 ;
G1 X60.00 E12.5  F1000.0 ; intro line
G1 X100.0 E14.0 F1000.0 ; intro line
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0

### Expected behavior

levelled printing

### Actual behavior

The nozzle starts to scratch the build plate when it moves to the right side of the flat bed.

### Version of Marlin Firmware

Marlin 2.1.2.1

### Printer model

QGL Core XY

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-05-19:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

fiveorders, 2023-05-19:
Hello I did as you requested:

Recv: >  X10.00 Y-4.00 Z10.00
Recv: <<< do_blocking_move_to  X10.00 Y-4.00 Z10.00
Recv: BLTouch DEPLOY requested
Recv: BLTouch from 90 to 10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to  X10.00 Y-4.00 Z10.00
Recv: >  X10.00 Y-4.00 Z10.00
Recv: <<< do_blocking_move_to  X10.00 Y-4.00 Z10.00
Recv: >>> Probe::run_z_probe  X10.00 Y-4.00 Z10.00
Recv: >>> Probe::probe_down_to_z  X10.00 Y-4.00 Z10.00
Recv: BLTouch DEPLOY requested
Recv: BLTouch from 10 to 10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to  X10.00 Y-4.00 Z10.00
Recv: >  X10.00 Y-4.00 Z-2.00
[...]
Recv: <<< do_blocking_move_to  X10.00 Y-4.00 Z-2.00
Recv: BLTouch STOW requested
Recv: BLTouch from 10 to 90
Recv: bltouch.stow_proc() end
Recv:   current_position= X10.00 Y-4.00 Z0.84 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X10.00 Y-4.00 Z0.84
Recv: <<< Probe::run_z_probe  X10.00 Y-4.00 Z0.84
Recv: >>> do_blocking_move_to  X10.00 Y-4.00 Z0.84
Recv: >  X10.00 Y-4.00 Z5.84
Recv: <<< do_blocking_move_to  X10.00 Y-4.00 Z5.84
Recv: measured_z: 0.84
Recv: <<< Probe::probe_at_point  X10.00 Y-4.00 Z5.84
Recv: >>> Probe::probe_at_point  X10.00 Y-4.00 Z5.84
Recv: ...(150.00, 45.00, raise, 0, probe_relative)
Recv:   current_position= X10.00 Y-4.00 Z5.84 :
Recv: >>> do_blocking_move_to  X10.00 Y-4.00 Z5.84
Recv: >  X150.00 Y-4.00 Z5.84
[...]
Recv: <<< do_blocking_move_to  X150.00 Y-4.00 Z5.84
Recv:   current_position= X150.00 Y-4.00 Z5.84 : Probe::set_deployed
Recv: deploy: 1
Recv: >>> Probe::run_z_probe  X150.00 Y-4.00 Z5.84
Recv: >>> Probe::probe_down_to_z  X150.00 Y-4.00 Z5.84
Recv: BLTouch DEPLOY requested
Recv: BLTouch from 90 to 10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to  X150.00 Y-4.00 Z5.84
Recv: >  X150.00 Y-4.00 Z-2.00
Recv: <<< do_blocking_move_to  X150.00 Y-4.00 Z-2.00
Recv: BLTouch STOW requested
Recv: BLTouch from 10 to 90
Recv: bltouch.stow_proc() end
Recv:   current_position= X150.00 Y-4.00 Z1.34 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X150.00 Y-4.00 Z1.34
Recv: <<< Probe::run_z_probe  X150.00 Y-4.00 Z1.34
Recv: >>> do_blocking_move_to  X150.00 Y-4.00 Z1.34
Recv: >  X150.00 Y-4.00 Z6.34
[...]
Recv: <<< do_blocking_move_to  X150.00 Y-4.00 Z6.34
Recv: measured_z: 1.34
Recv: <<< Probe::probe_at_point  X150.00 Y-4.00 Z6.34
Recv: >>> Probe::probe_at_point  X150.00 Y-4.00 Z6.34
Recv: ...(290.00, 45.00, raise, 0, probe_relative)
Recv:   current_position= X150.00 Y-4.00 Z6.34 :
Recv: >>> do_blocking_move_to  X150.00 Y-4.00 Z6.34
Recv: >  X290.00 Y-4.00 Z6.34
Recv: <<< do_blocking_move_to  X290.00 Y-4.00 Z6.34
Recv:   current_position= X290.00 Y-4.00 Z6.34 : Probe::set_deployed
Recv: deploy: 1
Recv: >>> Probe::run_z_probe  X290.00 Y-4.00 Z6.34
Recv: >>> Probe::probe_down_to_z  X290.00 Y-4.00 Z6.34
Recv: BLTouch DEPLOY requested
[... 191 lines cut ...]
Recv: >>> do_blocking_move_to  X290.00 Y241.00 Z4.28
Recv: >  X290.00 Y241.00 Z10.00
[...]
Recv: <<< do_blocking_move_to  X290.00 Y241.00 Z10.00
Recv: BLTouch STOW requested
Recv: BLTouch from 90 to 90
Recv: bltouch.stow_proc() end
Recv: >>> do_blocking_move_to  X290.00 Y241.00 Z10.00
Recv: >  X290.00 Y241.00 Z10.00
Recv: <<< do_blocking_move_to  X290.00 Y241.00 Z10.00
Recv:   current_position= X290.00 Y241.00 Z10.00 : > probing complete
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.837 +1.340 +1.923
Recv:  1 +0.369 +0.424 +0.550
Recv:  2 +0.000 -0.436 -0.721
Recv: 
Recv: >>> set_bed_leveling_enabled  X290.00 Y241.00 Z10.00
Recv:   current_position= X290.00 Y241.00 Z10.00 : Leveling OFF
Recv:   current_position= X290.00 Y241.00 Z10.00 : sync_plan_position
Recv:   current_position= X290.00 Y241.00 Z10.00 : Leveling ON
Recv: <<< set_bed_leveling_enabled  X290.00 Y241.00 Z10.00
[...]
Recv: <<< G29  X290.00 Y241.00 Z10.00
Recv: ok

fiveorders, 2023-05-20:
My problem is Slanting Bed Leveling. The nozzle starts to scratch the bed gradually when it moved to the right.

fiveorders, 2023-05-21:
Please see [https://github.com/MarlinFirmware/Marlin/issues/25866](url)

fiveorders, 2023-05-21:
https://github.com/MarlinFirmware/Marlin/issues/25866

Updated the issue as above

closed 2023-05-21 by fiveorders

classicrocker883, 2023-05-25:
sounds like the gantry sags and it's a hardware issue. usually having a Dual Z lead screw and motor solves this problem. simple easy upgrade.

fiveorders, 2023-05-25:
it was the X axis, slightly twisted so when the sensor at the back right side of the bed raises. When the sensor at the front left side raises.