MarlinFirmware/Marlin #25859: Bed crashes into hot end went print finishes

nikki-reprap, 2023-05-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have just upgraded to the latest release from 2.1.1 and when a print completes the bed moves up into the print when it should move down. My machine is a ratrig v3  and my mainboard is the octopus v1.1

### Bug Timeline

after upgrading

### Expected behavior

to moved the bed down into the hot end when the print completes 

### Actual behavior

It moved the bed up into the hot end when the print completes 

### Steps to Reproduce

Finish a print

### Version of Marlin Firmware

2.1.2.1

### Printer model

ratrig v3

### Electronics

octopus v1.1

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/11520842/config.files.zip)

thisiskeithb, 2023-05-19:
Can you also attach the `*.gcode` file that causes this issue?

EvilGremlin, 2023-05-19:
Duplicate of #25854

nikki-reprap, 2023-05-20:
[3DBenchy.zip](https://github.com/MarlinFirmware/Marlin/files/11521148/3DBenchy.zip)

closed 2023-05-20 by ellensp