MarlinFirmware/Marlin #25861: [BUG] JyersUI won't compile -- `corner_pos` not useable in constexpr?

classicrocker883, 2023-05-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

```
Marlin\src\lcd\e3v2\jyersui\dwin.cpp: In static member function 'static void CrealityDWINClass::Menu_Item_Handler(uint8_t, uint8_t, bool)':
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:1411:131: error: the value of 'corner_pos' is not usable in a constant expression
 1411 |         constexpr float probe_x_min = _MAX(0 + corner_pos, X_MIN_POS + probe.offset.x, X_MIN_POS + PROBING_MARGIN) - probe.offset.x,
      |                                                                                                                                
   ^
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:205:7: note: 'corner_pos' was not declared 'constexpr'
  205 | float corner_pos;
      |       ^~~~~~~~~~
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:1412:154: error: the value of 'corner_pos' is not usable in a constant expression
 1412 |                         probe_x_max = _MIN((X_BED_SIZE + X_MIN_POS) - corner_pos, X_MAX_POS + probe.offset.x, X_MAX_POS - PROBING_MARGIN) - probe.offset.x,
      |                                                                                                                                
                          ^
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:205:7: note: 'corner_pos' was not declared 'constexpr'
  205 | float corner_pos;
      |       ^~~~~~~~~~
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:1413:131: error: the value of 'corner_pos' is not usable in a constant expression
 1413 |                         probe_y_min = _MAX(0 + corner_pos, Y_MIN_POS + probe.offset.y, Y_MIN_POS + PROBING_MARGIN) - probe.offset.y,
      |                                                                                                                                
   ^
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:205:7: note: 'corner_pos' was not declared 'constexpr'
  205 | float corner_pos;
      |       ^~~~~~~~~~
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:1414:154: error: the value of 'corner_pos' is not usable in a constant expression
 1414 |                         probe_y_max = _MIN((Y_BED_SIZE + Y_MIN_POS) - corner_pos, Y_MAX_POS + probe.offset.y, Y_MAX_POS - PROBING_MARGIN) - probe.offset.y;
      |                                                                                                                                
                          ^
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:205:7: note: 'corner_pos' was not declared 'constexpr'
  205 | float corner_pos;
      |       ^~~~~~~~~~
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:1437:48: error: expected ';' at end of member declaration
 1437 |                 constexpr struct { xy_pos_t p, ProbePtRaise r } points[] = {
      |                                                ^~~~~~~~~~~~
```

<br>

### Steps to Reproduce

1. Enable `DWIN_CREALITY_LCD_JYERSUI`
2. Enable usable board and chip. ex. Creality V4 | V422 | V427 & STM32F103RE_creality in platformio.ini
3. Enable BL-Touch
4. Enable UBL, or Bilinear Bed Leveling

### Version of Marlin Firmware

latest bugfix-2.1.x

### Printer model

Ender-3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-05-21:
Introduced here https://github.com/MarlinFirmware/Marlin/pull/25631

classicrocker883, 2023-05-22:
[MarlinConfigs-JyersUI.zip](https://github.com/MarlinFirmware/Marlin/files/11525973/MarlinConfigs-JyersUI.zip)
 
any other config like for an Ender 3 would work, just enable **JyersUI**

jamespearson04, 2023-06-02:
Afraid I don't have the hardware either. You should open a pull request referencing this issue, and hopefully some of the others here would be kind enough to test it out.

classicrocker883, 2023-06-02:
post whatever code changes you do, I'll be able to test just let me know what to look for.

tombrazier, 2023-06-03:
@classicrocker883 Do you know how to get the changes from my branch at https://github.com/tombrazier/Marlin/tree/25861? Or just download the whole image from https://github.com/tombrazier/Marlin/archive/refs/heads/25861.zip.

thinkyhead, 2023-06-11:
This seems to be solved. Marlin build completes with the provided configs once they are updated (`EITHER` -> `ANY`, `BOTH` -> `ALL`) and repaired (e.g., `HEATER_0_MAXTEMP 305`).

closed 2023-06-11 by thinkyhead